tzapu/WiFiManager #300: How change ESP123456 to MyName123456?

glowlabs, 2017-01-25:
Is it possible to change the "ESP" text in the autoConnect AP to a custom value?

skx, 2017-02-01:
See the `README` file.

If you call the code like so, then it will use the name you've supplied:
 
        WiFiManager wifiManager;
        wifiManager.autoConnect("TRAM-TIMES" );

In my case that creates a WiFi network called `TRAM-TIMES`.

glowlabs, 2017-02-07:
Sorry, I wasn't clear in my post. I would like to have my custom name + chipID.

Something like "MySSID-120548".

Is that possible?

skx, 2017-02-07:
Yes.

The default value comes [from this code](https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L134).  You could use something similar like so:

      // Create a custom SSID, with chip ID appended
      String ssid = "MySSID-" + String(ESP.getChipId());

      // Start up the manager, using that string.
      WiFiManager wifiManager;
      wifiManager.autoConnect(ssid.c_str());

If this helps please do close the issue :)

glowlabs, 2017-02-07:
Perfect, thats exactly what I needed. Thanks!

closed 2017-02-07 by glowlabs