tzapu/WiFiManager #307: Current parameters as default?

MarcFinns, 2017-02-06:
Hello again,
In my implementation I have a number of custom parameters (e.g. mqtt server etc).
When moving the device to a different network, the config page forces to re-enter all of them and not only the new SSID/password.
Is there a way to prompt the user with the config page with the current parameter set already populated?
thanks!

tzapu, 2017-02-07:
hi,

using the code in the examples i provided that should not be the case.

i m currently playing with a 9 custom param sketch and the only thing i need to reenter are the ssid and pass.

cheers

MarcFinns, 2017-02-07:
Do you mean that the HTML page shows the current values as default? Or that if you submit them blank the old ones are retained?
Thanks.

tzapu, 2017-02-07:
that the exemples are loading the current values as defaults for the web portal, so the webportal should show the current saved values in the interface

MarcFinns, 2017-02-07:
Ok then, will check again. Thanks!

closed 2018-01-26 by tablatronix