tzapu/WiFiManager #308: configPortalTimeout improvements

tablatronix, 2017-02-07:
only timeout configportal if ap has no client

tablatronix, 2017-02-08:
pr 
https://github.com/tzapu/WiFiManager/pull/310

tablatronix, 2017-02-08:
My fix was simple and to keep configportal open if a client was connected to AP.

This avoids dealing with last request timestamp bumping and all the problems that creates and fails to solve for.

tzapu, 2017-02-08:
in my mind, the timeout should be paused or reset if there is a client connected to the ap or maybe just increase the timeout interval with 30 seconds (for example) each time the loop happens, it has not timed out yet and there is someone connected... this should give in effect a timeout that will happen 30 seconds + whatever was left of the timeout after the client disconnected...

a bit hacky maybe, but not bloated i think

tablatronix, 2017-02-08:
So just add a static global timeoutbuffer variable and add it to the timeout calculations ?

tablatronix, 2017-02-08:
hows this?

```
boolean WiFiManager::configPortalHasTimeout(){
    if(_configPortalTimeout == 0 || wifi_softap_get_station_num() > 0){
      _configPortalStart = millis(); // kludge, bump configportal start time to skew timeouts
      return false;
    }
    return (millis() > _configPortalStart + _configPortalTimeout);
}
```

tzapu, 2017-02-08:
that looks brilliant to me, thank you very much

tablatronix, 2017-02-08:
pushed to PR

tablatronix, 2017-02-08:
In the future maybe implement a keep alive headers or a microajax keepalive

closed 2017-02-08 by tzapu

tablatronix, 2017-02-20:
It looks there might be a problem with this solution.

Depending on how the client disconnects, wifi_softap_get_station_num continues to report client connected.

tablatronix, 2017-02-21:
This seems to be caused as with a ton of other problems, by bad sta connect while trying to run softap, it seems it causes issues with dhcp as well.

I was able to stabilize this a bit by `WiFi.disconnect(true);` before enabling softap if not connected.
setting `mode(WIFI_AP)` was not enough.

FernandoGarcia, 2017-04-06:
Hi @tablatronix @tzapu!

After this update the portal never goes to timeout even without any client.

Best regards.

FernandoGarcia, 2017-04-06:
I'm sorry was my mistake I don't know the reason but I changed the timeout from seconds to milliseconds. :D

tablatronix, 2018-02-08:
added webclient checking to bump timeouts, good enough until I decide to add js and ajax keepalives.

1c4011bfb4debe15952239fa6861b4e04cebba4f

commit 3135bc5e (2017-02-08): Merge pull request #310 from tablatronix/feature_308_configportaltimeouts

Feature 308 configportaltimeouts
fixes #308 
thank you very much