tzapu/WiFiManager #309: ass support for IEEE 802.1x

CWempe, 2017-02-07:
It would be great if WiFiManager would support authentication via [IEEE 802.1x](https://en.wikipedia.org/wiki/IEEE_802.1X).

In some universities there is no "normal" WiFi and you need to connect via 802.1x.
This would make developing in education much easier.

I understand this is supported with the new SDK 2.0.0 for esp8266.
Which is available for Arduino, I think.
https://github.com/esp8266/Arduino/commit/ae13809c8184300aab9e3f09ef23af23d936b7ee

tablatronix, 2017-02-08:
no where near stable though stable is still SDK 1.5.3
no milestone either, i have not even touched master in months, how stable is it ?

CWempe, 2017-02-08:
I have no idea if the new sdk is stable or not.
I just read somewhere that 802.1x is (or will be) supported with sdk 2.0.0.

l00mi, 2018-03-28:
Being new to this party I am wondering if the SDK 2.0.0 has matured and if 802.1x could be considered?

bkrajendra, 2019-09-26:
Im testing WPA2 PEAP with following code. will let you know if it works:

```
#include <ESP8266WiFi.h>

extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
}

// SSID to connect to
static const char* ssid = "IOCARE_NEW";
// Username for authentification
static const char* username = "myraddisuser";
// Password for authentication
static const char* password = "rapass123456";

void setup(){
 // WPA2 Connection starts here
  // Setting ESP into STATION mode only (no AP mode or dual mode)
    wifi_set_opmode(STATION_MODE);
    struct station_config wifi_config;
    memset(&wifi_config, 0, sizeof(wifi_config));
    strcpy((char*)wifi_config.ssid, ssid);
    wifi_station_set_config(&wifi_config);
    wifi_station_clear_cert_key();
    wifi_station_clear_enterprise_ca_cert();
    wifi_station_set_wpa2_enterprise_auth(1);
    wifi_station_set_enterprise_identity((uint8*)username, strlen(username));
    wifi_station_set_enterprise_username((uint8*)username, strlen(username));
    wifi_station_set_enterprise_password((uint8*)password, strlen(password));
    wifi_station_connect();
  // WPA2 Connection ends here
}
```

bkrajendra, 2019-09-27:
Tried all day long... but no success. getting error as follows while connecting:

```
Welcome to WPA2 Eneterprise Test
15:45:26.038 -> 1
15:45:26.072 -> WPA2 ENTERPRISE VERSION: [v2.0] disable
15:45:26.072 -> WPA2 ENTERPRISE VERSION: [v2.0] enable
15:45:26.072 -> scandone
15:45:26.072 -> 
15:45:26.072 -> Waiting for connection and IP Address from DHCP
15:45:26.106 -> wifi evt: 8
15:45:26.106 -> wifi evt: 2
15:45:28.096 -> .scandone
15:45:28.923 -> state: 0 -> 2 (b0)
15:45:28.923 -> state: 2 -> 3 (0)
15:45:28.923 -> state: 3 -> 5 (10)
15:45:28.923 -> add 0
15:45:28.923 -> aid 7
15:45:28.923 -> cnt 
15:45:29.024 -> Method private structure allocated failure
15:45:35.434 -> .EAP-PEAP: received 53 bytes encrypted data for Phase 2
15:45:35.606 -> EAP-PEAP: received Phase 2: code=1 identifier=145 length=5
15:45:35.606 -> EAP-PEAP: Phase 2 Request: type=1
15:45:35.606 -> .EAP-PEAP: received 85 bytes encrypted data for Phase 2
15:45:35.674 -> EAP-PEAP: received Phase 2: code=1 identifier=146 length=38
15:45:35.708 -> EAP-PEAP: Phase 2 Request: type=26
15:45:35.708 -> EAP-PEAP: Selected Phase 2 EAP vendor 0 method 26
15:45:35.708 -> EAP-MSCHAPV2: RX identifier 146 mschapv2_id 146
15:45:35.708 -> EAP-MSCHAPV2: Generate Challenge Response
15:45:35.708 -> .EAP-PEAP: received 101 bytes encrypted data for Phase 2
15:45:35.776 -> EAP-PEAP: received Phase 2: code=1 identifier=147 length=57
15:45:35.776 -> EAP-PEAP: Phase 2 Request: type=26
15:45:35.810 -> EAP-MSCHAPV2: RX identifier 147 mschapv2_id 146
15:45:35.810 -> EAP-MSCHAPV2: failure message: '' (retry allowed, error 691)phase 2 response failure
15:45:35.810 -> ..pm open,type:2 0
```
Searched a lot on various forums, tried many options. Currently using latest commit from master.
tried all possible way given on forums but no success.
I tried all this without using WiFiManager.
And im using static IP.

Im currently doing this setup for a Industry environment, and I guess i need to conclude that ESP8266/ESP32 are not made for Industry.

Most Industry has this compliance to use WPA2-Enterprise security in their networks.
So in short we can not sell any solution to Industry based on ESP.
If anyone tested please help. I understand this is not place to ask help... but eventually we all come at this place for wifi need. So any one got any success will be good to know.

tablatronix, 2019-09-27:
You running wireshark on this ?

bkrajendra, 2019-09-27:
No chance of any sniffing. They have very tight norms for using any third party software inside premises.
As I confirmed from IT person (who has very limited knowledge of all this RADIUS stuff) about the error that i got regarding EAP-MSCHAPv2 which is mostly due to settings in RADIUS serve. As he said its properly set to use MSCHAPv2.
As read on some forum about this that some server use MSCHAPv1 hence this error is shown.

Still I will not give up. today I'll try to setup my own RADIUS server using Raspberry Pi and test this flow.

tablatronix, 2019-09-27:
Yeah I googled

> 15:45:35.810 -> EAP-MSCHAPV2: failure message: '' (retry allowed, error 691)phase 2 response failure

and found a few confusing posts

kotelmach, 2020-10-30:
Has anyone got this working yet?

tablatronix, 2020-10-31:
afaik no one has looked into this any further
I guess we can revisit it has been awhile, not sure if there has been any development done

kumpakan, 2020-12-02:
With this code, I can connect to university network I work.

#include <ESP8266WiFi.h>

extern "C" {
  #include "user_interface.h"
  #include "wpa2_enterprise.h"
}

// SSID to connect to
static const char* ssid = "";
// Username for authentification
static const char* username = "";
// Password for authentification
static const char* password = "";

void setup() {

  // Setting ESP into STATION mode only (no AP mode or dual mode)
  wifi_set_opmode(STATION_MODE);
  struct station_config wifi_config;
  memset(&wifi_config, 0, sizeof(wifi_config));
  strcpy((char*)wifi_config.ssid, ssid);
  wifi_station_set_config(&wifi_config);
  wifi_station_clear_cert_key();
  wifi_station_clear_enterprise_ca_cert();
  wifi_station_set_wpa2_enterprise_auth(1);
  wifi_station_set_enterprise_username((uint8*)username, strlen(username));
  wifi_station_set_enterprise_password((uint8*)password, strlen(password));
  wifi_station_connect();

  // Wait for connection AND IP address from DHCP
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  // Now we are connected
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  
}

ildarlomov, 2021-01-22:
> With this code, I can connect to university network I work.
> 
> ```c++
> #include <ESP8266WiFi.h>
> 
> extern "C" {
>   #include "user_interface.h"
>   #include "wpa2_enterprise.h"
> }
> 
> 
> // SSID to connect to
> static const char* ssid = "";
> // Username for authentification
> static const char* username = "";
> // Password for authentification
> static const char* password = "";
> 
> void setup() {
> 
>   // Setting ESP into STATION mode only (no AP mode or dual mode)
>   wifi_set_opmode(STATION_MODE);
>   struct station_config wifi_config;
>   memset(&wifi_config, 0, sizeof(wifi_config));
>   strcpy((char*)wifi_config.ssid, ssid);
>   wifi_station_set_config(&wifi_config);
>   wifi_station_clear_cert_key();
>   wifi_station_clear_enterprise_ca_cert();
>   wifi_station_set_wpa2_enterprise_auth(1);
>   wifi_station_set_enterprise_username((uint8*)username, strlen(username));
>   wifi_station_set_enterprise_password((uint8*)password, strlen(password));
>   wifi_station_connect();
> 
>   // Wait for connection AND IP address from DHCP
>   while (WiFi.status() != WL_CONNECTED) {
>     delay(500);
>     Serial.print(".");
>   }
> 
>   // Now we are connected
>   Serial.println("");
>   Serial.println("WiFi connected");  
>   Serial.println("IP address: ");
>   Serial.println(WiFi.localIP());
> }
> 
> void loop() {
>   
> }
> ```

could you please add some more details? which model of the hardware? which version of esp8266 lib?

I've searched the implementation of bool wifi_station_connect(void) mentioned above and not found one in the source. In particular, there are no source files near the header in which the function definition is located ./Arduino/tools/sdk/include/user_interface.h Guess this work in progress.

ildarlomov, 2021-01-22:
what I've found is that on esp32 PEAP+mSCHAPv2 works and it's enough for me
hope it helps  some of the enthusiasts https://github.com/espressif/arduino-esp32/issues/160
I also found that wpa2 enterprise is not currently possible on esp8266

ernestocurty, 2021-07-27:
Hi all,
I am also interested in getting the WPA2 enterprise working. I am developing a scientific project (automated bioreactor for tissue engineering), and support for IEEE 802.1x would be highly desirable since most academic environments use this type of authentication.

I tested the code posted on  [this ](https://gist.github.com/Matheus-Garbelini/2cd780aed2eddbe17eb4adb5eca42bd6) discussion, which is very similar to the code posted above by @bkrajendra and **kumpakan**, and I was able to successfully connect to my institute network, which uses the following configuration:

> Security: wpa2-enterprise
> Authentication: PEAP
> No CA certificate
> MsCHAPv2

@ildarlomov, for this test, I used a generic/Chinese esp8266 (esp01) module from amazon. I also used Arduino IDE with the ESP8266 core, version 3.0.1
The mentioned headers in the code are part of the core. I located them inside the esp8266 core folder: `/packages/esp8266/hardware/esp8266/3.0.1/include`

Best,
Ernesto