tzapu/WiFiManager #311: client limits

tablatronix, 2017-02-08:
might not be a good idea, but was thinking it might be wasted as globals.

- [ ] ap user limit
- [ ] webserver user limit

This will have to be an optional limit, and it might cause issues if sessions do not close out and a user is having issues. But it might be desirable to not allow users on ap and webserver when someone is already configuring, in case not using passwords, prevent abuse ?

tzapu, 2017-02-08:
it's a good idea, but i think the question really is, is it worth it?
configuration is really done quite fast and people should be using a password to prevent abuse so i'm thinking there are probably other issues to tackle before this becomes a real issue

tablatronix, 2017-02-08:
Also worth noting the web server only handles 4 sockets.