tzapu/WiFiManager #312: softap left on

tablatronix, 2017-02-08:
#216 
#267 

probably more

When configportal is abandoned, ap mode is left persistent,

tzapu, 2017-02-08:
for now it's probably worth just going to WIFI_AP and then back to WIFI_STA rather than AP+STA it goes to now.

it was a half assed attempt to do something else i think

tablatronix, 2017-02-08:
Easiest fix is to make sure to turn off AP when timeout of configportal, or else it gets stuck on every time.
Also worth noting that switching modes alot may cause flash wear.

tablatronix, 2017-02-08:
man github is being slow today, could not even see or merge updates.

pr on its way

tablatronix, 2017-02-08:
best solution is to handle this directly via SDK, no flash involved
`wifi_set_opmode_current`
>Set Wi-Fi working mode to Station mode, SoftAP or Station + SoftAP, and do not
update flash.

That or toggle persistent before and after changing modes.

tablatronix, 2017-02-08:
Testing good so far

```
/**
 * toggle mode, non persistant
 * @param m WiFiMode_t
 */
bool WiFiManager::wifimode(WiFiMode_t m) {
    DEBUG_WM(F("changing wifi mode non persistant"));
    if(wifi_get_opmode() == (uint8) m) {
        return true;
    }

    bool ret = false;

    ETS_UART_INTR_DISABLE();
    ret = wifi_set_opmode_current(m);
    ETS_UART_INTR_ENABLE();

    return ret;
}
```
Ideally we would be able to rework the library to not step on users configs.

do a `wifi_station_get_config` save state locally, switch to lib ap, finish,  restore users ap.
or when in doubt reboot works now since not overwriting flash.

tablatronix, 2017-02-10:
It is also worth noting that even though opmode wont be saved when you start the softap the conf is always saved.

neilyoung, 2017-02-11:
How do you  ETS_UART_INTR_ENABLE(); after return?

tablatronix, 2017-02-11:
huh?

closed 2022-01-21 by tablatronix