tzapu/WiFiManager #316: Weird behaviour when adding custom head.

ArnieX, 2017-02-12:
Hi @tzapu,

first of all, thanks for great library. I hope you could point me into right direction, I have quite altered your original design using custom head, added image to h1 instead of text. Hide h3, etc.

But this is what happens to me multiple times. Sometimes it works perfectly, other times it do not and I can't see any scheme or how to reproduce this behaviour.

This is how it should look like:
[image]

This is what happens:
[image]
[image]
[image]

Sometimes the main screen works but when I click any section it opens empty:
[image]

Any idea? Thanks for help.

BR,

Martin

nhaduy99, 2017-04-08:
Similar issue, I have tested that, in my case the scanning network function (WiFi.scanNetworks();) case a bit long time and this make the web crash or appears nothing.
Looking for solution.........
Regards

ArnieX, 2017-04-08:
Thanks, let me know if you find anything.

closed 2018-01-26 by tablatronix

tablatronix, 2018-01-27:
#400