tzapu/WiFiManager #321: Why I can't configure WiFi settings after a timout?

FernandoGarcia, 2017-02-23:
Hi!

After a timeout I can see the AP SSID and connect to it but I can't open the page (192.168.4.1) to configure.

The browser can't reach the page but local IP (192.168.4.2) and gateway (192.168.0.1) on laptop are fine.

This behavior is expected to this library?

I'm using this code:

```
void configWifi()
{
  WiFiManager wifiManager;
  wifiManager.setTimeout(180); // Entra em modo AP por 3 minutos.
  wifiManager.setDebugOutput(true); // Desativa mensagens de debug da biblioteca.

  if (!wifiManager.autoConnect("ESP8266", CLIENT_ID))// Nome e senha do ponto de acesso.
  {
#ifdef DEBUG
    Serial.println(F("Nao foi possivel conectar a rede WiFi, por favor  conecte-se ao ponto de acesso e tente novamente"));
#endif
  }
  else
  {
#ifdef DEBUG
    Serial.println(F("WiFi conectado"));
    Serial.print(F("Endereco IP: "));
    Serial.println(WiFi.localIP());
#endif
  }
}
```
I'm sorry if this question was answered previously but I can't find it.

Best regards.

closed 2018-01-26 by tablatronix

FernandoGarcia, 2018-01-26:
@tablatronix Why you are closing the issue without an answer?

mindforger, 2018-01-26:
@FernandoGarcia he is closing old issues, because there were WAY too much open issues :) nothing personal