tzapu/WiFiManager #325: Can saved wifi credentials be cleared using the reset button.

FaizB, 2017-03-02:
Hi I am new to the ESP8266

I am using the wifiManager to give me an access point to configure my wifi credentials. 
I was wondering without using ( wifiManager.resetSettings(); ) explicitly. So every time I have to input my wifi credentials is there a method where I can use the RESET button on the nodeMCU to initialize (wifiManager.resetSettings();), so I can change wifi credentials whenever I choose or send it into AP mode when the RESET button is pressed.

tusharpc, 2017-03-03:
look for KEn taylor's fork here for his version of the wifi manager

closed 2018-01-26 by tablatronix

trungkiendt9, 2018-03-16:
//Reset mạng wifi đã lưu tên
        int reset_level = analogRead(A0);
        Serial.print("reset_level: ");
        Serial.println(reset_level);

        if (reset_level >500) {
                WiFi.disconnect(true);
                delay(2000);
                ESP.reset();
        }

saleemnabaki, 2019-07-02:
trungkiendt9  Thank you so much .. Work very well