tzapu/WiFiManager #328: Q: EEPROM storage for MQTT access data

aseerig, 2017-03-04:
I want to put the MQTT access data into the EEPROM, without an additional SD card. I'm not a good programmer and I'm looking for a good example. I am afraid that I will overwrite the EEPROM data of the WiFiManager.

tzapu, 2017-03-04:
hi

see my sonoff boilerplate, it should have all you need and then some

cheers

aseerig, 2017-03-07:
Hello, the mqtt implementation has worked well.
I wanted to add the "alias" as another value, but I do not get it. I read with server and alias always the same value.
What the hell does the value "#define EEPROM_SALT 12667"

In principle, the node should use the ChipID as a prefix for the topic. Via the alias (example: "bathroom") I can set the location over the Frontend.

My current code is here:
Https://github.com/aseerig/SensorNode.git 

Can you give me a tip for implementing another value?

merlokk, 2017-04-19:
https://github.com/merlokk/SmartHome/blob/master/ESP8266EASTRON/ESP8266EASTRON.ino#L301

closed 2017-08-27 by tablatronix