tzapu/WiFiManager #329: Crash when using WiFiManagerParameter

daniftodi, 2017-03-05:
Hi,

When using WiFiManager with custom parameters, when connecting to AP and trying to access configuration page, my devices restarts ( ESP-12-F ), here is the serial output:

```
*WM: BM29
*WM: -93
 
Exception (28):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000014d depc=0x00000000
 
ctx: cont
sp: 3ffefe10 end: 3fff01a0 offset: 01a0
```
BM29 is the last Wifi network indentified by WiFiManager and I think before this, custom parameters are rendered.

Here is the source code that I'm using:

```
    char mqtt_server[40] = "mqtt server";
    char mqtt_port[6] = "8080";
    char mqtt_topic[63] = "mqtt/topic";

    WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
    WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);
    WiFiManagerParameter custom_mqtt_topic("topic", "mqtt topic", mqtt_topic, 63);

    wifiManager.addParameter(&custom_mqtt_server);
    wifiManager.addParameter(&custom_mqtt_port);
    wifiManager.addParameter(&custom_mqtt_topic);
    wifiManager.setSaveConfigCallback(saveConfigCallback);
    if(!this->wifiManager.autoConnect(WIFI_SSID, WIFI_PASS)){
      ESP.reset();
      delay(5000);
    }
```

Help me please figure what is wrong with my code and what should I do.

rkoptev, 2017-09-09:
Having the same issue.

It lloks like WiFiManagr consumes too much RAM:

Fatal exception 29(StoreProhibitedCause):
epc1=0x4000e1c3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: sys 
sp: 3ffffc80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe20:  ffffff80 3ffef190 00000000 00000001  
3ffffe30:  4021dc5d 4021db9a 3fff2dbc e0031035  
3ffffe40:  00000018 3ffefb40 3ffefabc 00000000  
3ffffe50:  4021dcf8 3ff20a00 3fff2dbc 3ffecc7e  
3ffffe60:  3ffecc64 0000000e 00000000 3ffecc64  
3ffffe70:  3ffecc7e 0000002c 00000008 401004f4  
3ffffe80:  3ffecc6e 3fff3344 4021aa4f 3ff20a00  
3ffffe90:  3fff2dbc 3fff4e54 00000015 4021a921  
3ffffea0:  402113a1 00000030 0000001a ffffffff  
3ffffeb0:  40211393 3fff13b0 40211418 3fffd9d0  
3ffffec0:  00000000 00000000 00000000 fffffffe  
3ffffed0:  ffffffff 3fffc6fc 00000001 3fff13b0  
3ffffee0:  00000000 3fffdad0 3fff13a8 00000030  
3ffffef0:  00000000 3fffdad0 3fff13a8 00000030  
3fffff00:  401068c4 00000000 00000007 401004d8  
3fffff10:  40221eb9 ffffffa7 00000000 00000000  
3fffff20:  4021a1b9 3ffecc64 3ffef190 3fff2dbc  
3fffff30:  00000000 3ffecc6e 3ffecc84 3ffeb534  
3fffff40:  40215624 00000000 3ffecc6e 3ffecc64  
3fffff50:  00000058 3ffeb534 3fff3344 3ffecc58  
3fffff60:  00000000 00000007 40215256 3ffef190  
3fffff70:  3ffecc58 3fffdcc0 3ffeb7f8 3fff3064  
3fffff80:  3ffecc74 00000040 00000000 3fff13b0  
3fffff90:  40214ba3 3fffdab0 00000000 402113a7  
3fffffa0:  3ffeb7f8 40000f49 3fffdab0 40000f49  
<<<stack<<<

Here is the output from ESP exception decoder:

tablatronix, 2017-09-09:
what branch ?

rkoptev, 2017-09-09:
I'm using ESP12E module from AI-Thinker. In Arduino IDE I selected NodeMCU 1.0(ESP-12E module)

tablatronix, 2017-09-09:
Oops I edited your post
What version branch of wifimanager?

rkoptev, 2017-09-09:
As mentioned in library.properties file: version=0.12

tablatronix, 2017-09-09:
Ok cause I think they all say the same
Making sure you are not using dev or kentaylor branch forks

Ill see if i can reproduce

rkoptev, 2017-09-13:
It seems that I managed to make some progress in solving this problem. I began to disable the libraries in my sketch in turn. WiFiManager started working correctly after I disabled SoftwareSerial library that I use to recieve data from GPS sensor.

I assumed that the matter is in timers or interrupts. Perhaps interrupts that are used in esp SoftwareSerial library to get data from the sensor prevent the work of the WiFiManager. I analyzed this library, and found out that when creating an softserial object, constructor calls enableRx() method that calls this:
`attachInterrupt(m_rxPin, ISRList[m_rxPin], m_invert ? RISING : FALLING);`

I can not be sure, but it seems to me that this was the main problem. In any case, I managed to solve this problem by calling enableRx(0) before starting WiFiManager.

tablatronix, 2017-09-13:
Also make sure you are never doing work inside an interrupt, instead use flag states or some other way,

closed 2018-01-26 by tablatronix