tzapu/WiFiManager #330: Repeated Smartconfig gives error

romuye123, 2017-03-09:
Following is my code for smartconfig. 
```
WiFi.mode(WIFI_STA);
    delay(1000);
    WiFi.beginSmartConfig();
    delay(1000);
    
    while (WiFi.status() != WL_CONNECTED) {
      counter++;
      delay(500);
      if(WiFi.smartConfigDone()){
          delay(10);
      }
      if(counter == 120){ 
            counter = 0;
            WiFi.stopSmartConfig();
            break;
      }
    }
```

If i try to do Smartconfig again and again(already connected to Wifi) gives following error:
```
there is no poison after the block. Expected poison address: 0x3fff47fe, actual data: 0x10 0x40 0x0 0x0
block start: 3fff47f4

```

romuye123, 2017-08-02:
@tzapu any update on this error. please help...

closed 2018-01-26 by tablatronix

tablatronix, 2018-01-27:
This is a esp platform issue, bug.