tzapu/WiFiManager #335: Static IP does not work after reboot

bobybobybob, 2017-03-21:
Hello,
the "AutoConnectWithStaticIP" Excampel does not work very good. 

Default IP in Arduinoscript is 192.168.1.50

When I set in webconfig (192.168.4.1) the  192.168.1.66 as new IP it is OK

But power off, it go back to 192.168.1.50

Look like a big Bug. I use ESP8266 EPS-12E and Arduino IDE 1.8.1

The "AutoConnectWithFSParametersAndCustomIP" works better but it contains some unknow MQTT??? Dont know what that is

--------------------

Next Problem

First use of Wifi Manager, it connect imediately with correct WLAN but i have never used wifi Manager before, only other ESP-Stuff with hardcoded WLAN SSID and Password.

-------------------

Maybee you have time to fix the excampel script. And I will pray for your soul. Alleluia Jesus loves ESP8266 too.

telanoc, 2017-03-30:
I'm not tzapu, but maybe I can answer your "next problem" part.  The ESP keeps track of the last AP and password that it connected to.  It can then connect back up to it easily.  WiFiManager comes into play when that "last-connected" value isn't set, or if that particular AP isn't around to connect to.

You can call `WiFi.disconnect(true);` once in your code to have the ESP wipe out its idea of that last connected AP (and disconnect you from it at the same time).  Then WiFiManager will kick in the next time you restart.

yovanfowdar, 2017-04-19:
bobybobybob , you are right this is exactly the same issue I am having, after setting the IP on the wifi manager page , everything ok until I reboot . I have tried with both block1 and block2, after reboot the IP address is not stored and used for the next connection. I hope a solution is found soon . Tzapu and guys working on this project , grateful if you could look into that . 
Thanks

tzapu, 2017-04-20:
hi guys, like the extra fs/eeprom parameters, you are responsible for storing the ip configuration somewhere and restoring it at next boot.

the sdk does not store it and wifimanager by itself, does not store anything.

i hope this helps
alex

yovanfowdar, 2017-04-20:
Thank you for your reply Tzapu , that shed more light , I guess we were all expecting that the entered static data  IP address, subnet mask and gateway were being stored on the fs  , which is not the case . This reply will surely help us in understanding  how the wifi manager for static configuration works. A nice enhancement would be having the static data be stored on the eprom. Cheers

bobybobybob, 2017-04-20:
Hello dear friends,
Can it be that the error is dependent on the eeprom module? With some ESP12E everything works wonderfully, with ESP-07 with start-up problems then OK, with ESP-01 not at all and with some ESP12E and an ESP12F it did not go either. The ESPs at 12E have different manufacturers of the Eeprom.

closed 2018-01-26 by tablatronix

bobybobybob, 2018-01-26:
tablatronix closed this 10 hours ago

have you fix the problem?

mrgadotti, 2021-03-08:
I did have the same problem. After change IP adress  on  from Web portal, it's work fine. After reboot, the board use the IP from sketch. The same issue happens on ESP-07 and ES-12E.