tzapu/WiFiManager #336: Unable to connect to a new AP if currently connected to the old AP

babitsky, 2017-03-22:
Hello,

I'm calling startConfigPortal on demand. 
The module is currently connected to AP "x".
Using the config portal, I select a new AP "y", with new credentials.
The module attempts to connect to the new AP "y", but "fails" (inappropriate optimization for this scenario) since it remains connected to "x":
```
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
```
New AP "y" and its credentials are therefore lost. 
Module remains connected to "x".

pbecchi, 2017-03-25:
I have exactly the same problem!
One I am connected to an AP I cannot change it with startconfigportal call!

telanoc, 2017-03-30:
I would add `WiFi.disconnect(true)` to force a disconnection (and clearing of saved credentials) before calling `startConfigPortal()`.

tzapu, 2017-03-30:
hi guys,
this is a bug in the library, the disconnect trick will work as a workaround 

cheers

mouk, 2017-04-18:
@tzapu has this bug been reported?

babitsky, 2017-04-18:
@telanoc  I'm not sure clearing the credentials is the best solution... At least not for my application.

My suggestion is to add a check in ```connectWifi()``` to make sure the "Already connected. Bailing out." code is not executed when connecting to a new SSID (the ```&& (WiFi.SSID() == ssid)``` part):

```
  //fix for auto connect racing issue
  if ((WiFi.status() == WL_CONNECTED) && (WiFi.SSID() == ssid)) {
    DEBUG_WM("Already connected. Bailing out.");
    return WL_CONNECTED;
  }
```

I plan to submit it as a pull request one day, when I learn how to do it :)
So far, it seems to have solved the problem for me completely.

commit d0e70fdd (2019-08-30): fixes #336
closed 2019-08-30 by tablatronix