tzapu/WiFiManager #350: Accespoint Config Mode does not work well

bobybobybob, 2017-04-07:
Hello, I have four eps12E. I put these in Accespoint Config mode to display Accespoint.

With two I can connect and adjust everything.
For the other two, a stable connection is not possible. The signal travels in a few seconds so I do not see the config page.

The script code is the same for all four. Also the same firmware. Identical current source.

In serial print, writes on the non workung one something like reset ets Jan 8 2013, rst cause: bootmode:
Wdt reset

Is this a Software Bug oder Hardware problem?

maxdd, 2017-04-14:
Have you checked your hw pins configuration of the second device?

javieryanez, 2017-09-21:
I have the same problem with a Witty Cloud (ESP-12F)

tablatronix, 2017-09-21:
Could be power supply could be corrupt memory try esptool erase flash?

bobybobybob, 2017-09-21:
> Could be power supply could be corrupt memory try esptool erase flash?
No, i have 8 ESP12E and F from diferent sellers. 3 are very stabile and the other are shit. 

Super Power 3.4V 3A, many Capacitators and that Pullup, Pulldown rubish for improve Stability helps not much. 

I found out that the instabile ESP need one connection to WLAN and after upload new script, the problem script work without problems and most stabile. It is maybe a Hardwarebug.

tablatronix, 2017-09-21:
What sketch sounds like

tablatronix, 2017-09-21:
what sketch ? Does the sketch cause the reset on purpose, if so it is probably this issue

#425

bobybobybob, 2017-09-21:
Excample

I buy a new RAW ESP12E or ESP12F. The lite one no wemos. I upload a script likte

in the loop
int blabla = 0
blabla = blabla + 1
serial.print blabla

it will count up maybe up to 2000 then the watchdog Reset come and it start again new

When i upload the same sketch again, it count rising. Maybe in first it count up to 5000, in the second upload it count only to 2000 and after some more most counting end ner 1000.

But if i upload a script that make hardcoded wlan connection and then overwrite it withh the counter script without wlan, it count up to endlees or one some ESP the watchdog again but in very higher numbers where a litte capacitator realy help stability. 

Is that ESP12 one time in his life connected with wlan, it is stability. Stability go away when flash firmware new + erase.

bobybobybob, 2017-09-21:
do you can read german? if yes i can write Deutsch

closed 2018-01-26 by tablatronix