tzapu/WiFiManager #351: Problem with WiFi.scanNetworks();

nhaduy99, 2017-04-08:
Hi,
I am using this library but sometimes the config AP portal doesn't show off when scanning the local wifi (page 192.168.4.1/?wifi). I check on the Serial Monitor:
*WM: Scan done
*WM: Yamaha
*WM: -60
*WM: My Dung_plus
*WM: -87
*WM: Sent config page

Sometimes the web appears with missing parts and the line with CTYPE html>. I searched that this wear word appears when the web is not fully HTML website. 

How do I fix this?

Best regards,
Seppy

svh1985, 2017-04-29:
I'm having the same issues; especially when there are a lot of wifi networks around.

tablatronix, 2017-08-27:
How many aps?
I have 25 and have never encountered this

svh1985, 2017-08-27:
I can test it when I'm back home, we have 40+ WiFi networks from all the neighbours at home.

tablatronix, 2017-08-27:
k
interestingly enough I am having a similar symptom unrelated to this issue

svh1985, 2017-08-27:
I'm guessing that it has something to do with the ESP running out of memory when it has to do all the "find & replace" actions during the HTML output generation.

tablatronix, 2017-08-27:
Yeah I am going to check the output and see whats mangled, oddly mine only occurs if i am not calling handlewifi fast enough, so there is some asynchronous problems with the server handling.

tablatronix, 2017-08-27:
_apname is const char and getting lost
refactoring in my dev branch required me to change this anyway for non blocking mode, and inadvertently fixes this bug also. Basically chanve _apname and _appassword to strings, nuch easier to deal with.

closed 2020-05-11 by tablatronix