tzapu/WiFiManager #352: The wifilibrary called many "Request redirected to captive portal"

nhaduy99, 2017-04-10:
Hi, I use the WifiManager lib with the example file of Autoconnect with FS parameter. Sometimes, when I click on "Configure Wifi" button, after scanning the SSID and the HTML seem stuck. I check on the Serial Monitor, the request redirected to captive portal appear so many time, and then handleRoot. 

*WM: Scan done
*WM: Yamaha
*WM: -61
*WM: DIRECT-tY-BRAVIA
*WM: -71
*WM: My Dung_plus
*WM: -88
*WM: My Phuong
*WM: -93
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root

Does anyone know what happens? Is it reset the wifi? What can I fix this. 
Sorry if this issue has solved but I haven't found the solution yet.

Thanks

tuxedo0801, 2017-06-01:
Happens to me as well ... very strange.

mnsk, 2017-07-19:
Having same issue. It works properly for the first time if I reset settings and format FS.

closed 2018-01-26 by tablatronix