tzapu/WiFiManager #355: Wifimanager with FS parameter - FS collapses and resets continuously

nhaduy99, 2017-04-15:
Hi, 
I have a problem with this, I used Wifimanager with FS parameter example. At first, everything is OK, then after a few days my ESP8266 WeMos D1 doesn't work as before, and it juts resets continously. I checked the Serial Monitor and this is the reason:
mounting FS...
mounted file system
reading config file
opened config file
{"table_ID":"hello","restaurant_ID":"seppy"}
parsed json

Exception (28):
epc1=0x4000bf0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff04b0 end: 3fff07b0 offset: 01a0

>>>stack>>>
3fff0650:  3fff07d4 3fff157c 3ffef6c0 402073c7  
3fff0660:  3ffe9180 00000000 000003e8 feefeffe  
3fff0670:  3fff153c 3fff1084 feefeffe feefeffe  
3fff0680:  feefeffe feefeffe feefeffe feefeffe  
3fff0690:  feefeffe feefeffe feefeffe feefeffe  
3fff06a0:  feefeffe feefeffe feefeffe feefeffe  
3fff06b0:  feefeffe feefeffe feefeffe feefeffe  
3fff06c0:  feefeffe feefeffe feefeffe feefeffe  
3fff06d0:  feefeffe feefeffe feefeffe feefeffe  
3fff06e0:  feefeffe feefeffe feefeffe feefeffe  
3fff06f0:  feefeffe feefeffe feefeffe feefeffe  
3fff0700:  feefeffe feefeffe feefeffe 3fff0724  
3fff0710:  3fff15a8 3fff159f feefef0a feefeffe  
3fff0720:  feefeffe 3ffe8870 feefeffe 3fff1f14  
3fff0730:  00000200 feefeffe feefeffe feefeffe  
3fff0740:  feefeffe feefeffe feefeffe feefeffe  
3fff0750:  feefeffe feefeffe feefeffe feefeffe  
3fff0760:  feefeffe feefeffe feefeffe feefeffe  
3fff0770:  feefeffe feefeffe feefeffe feefeffe  
3fff0780:  feefeffe feefeffe feefeffe 3ffef77c  
3fff0790:  3fffdad0 00000000 3ffef774 4020e6b8  
3fff07a0:  feefeffe feefeffe 3ffef790 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

I dont know what happened. How can I solve this.

Thanks

spudwebb, 2017-08-30:
I'm having the same kind of issue.
Did you find the cause?

tablatronix, 2017-08-30:
This is usually a memory access error

Do you have a sample sketch?

spudwebb, 2017-08-31:
Turns out it was caused by a non initialized pointer in my code. 
Everything is working as expected now
Thanks!

rkoptev, 2017-09-09:
Here is my sketch:
https://pastebin.com/FCZs3k4S

Asahajit, 2018-01-23:
I am also having the same issue. Tried 2nd and 3rd example it's failing.

Error 👎 

mounting FS...
mounted file system
reading config file
opened config file
{"apiKey":"123456","channelId":"UCezJOfu7OtqGzd5xrP3q6WA"}
parsed json

Exception (28):
epc1=0x40212442 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0b90 end: 3fff0ea0 offset: 01a0

>>>stack>>>
3fff0d30:  0000003a 3fff2f74 3ffefdb0 40207f76  
3fff0d40:  3ffe95f8 00000000 000003e8 feefeffe  
3fff0d50:  3fff2f1c 3fff2f5c feefeffe feefeffe  
3fff0d60:  feefeffe feefeffe feefeffe feefeffe  
3fff0d70:  feefeffe feefeffe feefeffe feefeffe  
3fff0d80:  feefeffe feefeffe feefeffe feefeffe  
3fff0d90:  feefeffe feefeffe feefeffe feefeffe  
3fff0da0:  feefeffe feefeffe feefeffe feefeffe  
3fff0db0:  feefeffe feefeffe feefeffe feefeffe  
3fff0dc0:  feefeffe feefeffe feefeffe feefeffe  
3fff0dd0:  feefeffe feefeffe feefeffe feefeffe  
3fff0de0:  feefeffe feefeffe feefeffe 3ffefdb0  
3fff0df0:  0000003a 3fff2fa5 feefef0a feefeffe  
3fff0e00:  feefeffe 3ffe8d10 feefeffe 3fff2fb4  
3fff0e10:  00000200 feefeffe feefeffe feefeffe  
3fff0e20:  feefeffe feefeffe feefeffe feefeffe  
3fff0e30:  feefeffe feefeffe feefeffe feefeffe  
3fff0e40:  feefeffe feefeffe feefeffe feefeffe  
3fff0e50:  feefeffe feefeffe feefeffe feefeffe  
3fff0e60:  00000000 feefeffe feefeffe feefeffe  
3fff0e70:  feefeffe feefeffe feefeffe 3ffefe70  
3fff0e80:  3fffdad0 00000000 3ffefe68 4020f6b0  
3fff0e90:  feefeffe feefeffe 3ffefe80 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
r$

tablatronix, 2018-01-23:
Can you post exact reproducable steps and code?

closed 2018-01-26 by tablatronix

Asahajit, 2018-01-26:
it worked and issue got fixed with "SPIFFS.format();" .