tzapu/WiFiManager #360: moving var to public section

dvarrel, 2017-04-24:
it would be appreciate to declare String _ssid and _psk in public section instead of private (WiFiManager.h).

then i can pick up these variable after successfully connected with portal and save them to eeprom

127 String _ssid = "";
128 String _pass = "";
move it to public section in WiFiManager.h

thank you

tzapu, 2017-04-25:
hi,

you don t need to go through wifimanager for that

just use
WiFi.SSID() and WiFi.psk()

cheers

dvarrel, 2017-04-25:
hi,
but i can't change it, no ?

tzapu, 2017-04-25:
you can change/save it with WIFi.begin(ssid, pwd) or whatever the syntax is. the standard esp8266 sdk functions

cheers

dvarrel, 2017-04-25:
yes, but i would like to change it before calling wifiManager.autoConnect(); and not WIFi.Begin()

closed 2017-08-27 by tablatronix