tzapu/WiFiManager #361: ESP-01 issues (Generic version)

klerone, 2017-04-28:
tzapu.

You made a great job with this library I use it with the ESP-12 like a charm, but I have problem with the ESP-01, for some reason the same sketch that works in the ESP-12 does not work in the ESP-01.

- ESP-01 does not run the Wifi Manager if does not find a know wifi network.
- The examples sketch can upload successfully (is not problem to upload the sketch).
- ESP-01 run my other sketches without problems.

I'm pretty sure in not my wiring to the ESP-01 since I made my own custom board and I set it up as a Socket server and works smoothly, I wonder if the ESP-01 is not supported. 

I have 5 ESP-01 (black color, say AI Thinker) and none of them can run the manager.

Any help is welcome.

tablatronix, 2017-08-27:
You provided no useful details

closed 2017-08-27 by tablatronix

marcblount, 2018-02-28:
Hi both,

Apologies for opening up a closed thread, however i have come across the same issue but unable to find a solution.

I am experiencing the same issue - wifimanager doesn't seem to run on the ESP-01.  I can get wifimanager to run well on my ESP8266 but none of my ESP-01 devices.

Do we know if this a known issue or limitation?

Regards

kosso, 2018-03-01:
I just got it working today on an ESP-01. (Black PCB)

Fabi12345, 2018-05-07:
Would you please provide your configuration? My ESP-01's don't work either. 
OnDemandConfigPortal is properly opening the mask, I can connect but have no internet access through it.

dontsovcmc, 2018-06-09:
Hi all!
My ESP-01 works perfect, but I see some strangers:
- one day MacOS don't want to show web page 192.168.4.1, curl responses fail. I open virtual windows and browser show 192.168.4.1 correctly (wtf?!)

@klerone provide your code, please