tzapu/WiFiManager #362: Content-length is not set properly

suculent, 2017-05-03:
Works on iPhone, but fails on certain Androids accessing the Captive Portal. Solution to fix this was straitforward... adding explicit Content-Length header before all ->send requests.

suculent, 2017-05-03:
Following solution works for me:

https://github.com/suculent/WiFiManager/commit/ada93b721a235f4bae379e1804c32d962541b992

scop, 2017-05-26:
I was bitten by this on Android 7.1.2, and the suggested change works for me too, please apply.

scop, 2017-07-10:
@suculent, could you submit a pull request to get this fix in? Thanks!

suculent, 2017-07-10:
I've already done this quite a while ago https://github.com/tzapu/WiFiManager/pull/363

tzapu, 2017-07-10:
thanks guys, merged

closed 2017-07-10 by tzapu

tablatronix, 2017-09-15:
The sdk seems to already send content-length headers, I wonder why android is so picky about the order maybe ?

suculent, 2017-09-15:
It did not happen only on Android, at least I'm not the Android case, we use iPhones and iPad Pro to test WiFiManager in AP mode. 

I need to find correct test case. After upgrading to latest WiFiManager I need to solve own issues first before going back to AP mode. 

Odesláno z iPhonu

tablatronix, 2017-09-15:
Maybe this changed in SDK at some point, shrug