tzapu/WiFiManager #367: WPS WorkaRound

Melcos1970, 2017-05-08:
I have made a small change to WiFiManager.cpp and .h to get a mid-term solution to the problem with WPS.
It's not the best way, but now you can use WPS if you are going to ..
I have no idea how to add suggestions to github, but I want to post them here so can add them if you want to use them.
The solution is actually simple .. I just add a button extra that makes a WPS call.

Changed:
Const char HTTP_PORTAL_OPTIONS [] PROGMEM = "<form action = \" / wifi \ "method = \" get \ "> <button> Configure WiFi </ button> </ form> <br/> <form action = \" / 0wifi \ "method = \" get \ "> <button> Configure WiFi (No Scan) </ button> </ form> <br/> <form action = \" / i \ "method = \" get \ "> <Button> Info </ button> </ form> <br/> <form action = \ "/ r \" method = \ "post \"> <button> Reset </ button> </ form> <br/> <Form action = \ "/ w \" method = \ "post \"> <button> WPS Setup </ button> </ form> ";

Inserted: lines +++
    Void handleReset ();
+++ void handleWPS ();
    Void handleNotFound ();

  Server-> on ("/ r", std :: bind (& WiFiManager :: handleReset, this));
+++ server-> on ("/ w", std :: bind (& WiFiManager :: handleWPS, this));
  // server-> on ("/ generate_204", std :: bind (& WiFiManager :: handle204, this)); // Android / Chrome OS captive portal check.
  Server-> on ("/ fwlink", std :: bind (& WiFiManager: handleRoot, this)); // Microsoft captive portal. Maybe not needed. Might be handled by notFound trades.

Inserted section / ++ ++ / after void WiFiManager :: handleReset () section
/ ++
Void WiFiManager :: handleWPS () {
  DEBUG_WM (F ("WPS Start"));

  String page = FPSTR (HTTP_HEAD);
  Page.replace ("{v}", "Info");
  Page + = FPSTR (HTTP_SCRIPT);
  Page + = FPSTR (HTTP_STYLE);
  Page + = _customHeadElement;
  Page + = FPSTR (HTTP_HEAD_END);
  Page + = F ("Module will start WPS in a few seconds.");
  Page + = FPSTR (HTTP_END);
  Server-> send (200, "text / html", page);

  DEBUG_WM (F ("Sent WPS page"));
  // WPS works only in STA (Station mode) only.
  WiFi.mode (WIFI_STA);
  delay (1000);
  startWPS ();
  delay (5000);
  ESP.reset ();
  delay (2000);
}
++ /

[WiFiManager_WPS_Files.zip](https://github.com/tzapu/WiFiManager/files/983665/WiFiManager_WPS_Files.zip)

bkrajendra, 2017-05-23:
The main problem with WPS that i found is that it does not retain its password on reset.
Did you get this issue?

Melcos1970, 2017-05-25:
No not with WiFiManager.. My problem is if the WPS fail to connect and the go back til normal state, that is why i make a reboot after using WPS.

tablatronix, 2017-08-27:
I know this is old but can you post your code in code blocks please use three ticks open and close

MarcUbb, 2023-01-22:
Hey, I think I found a solution to the WPS problem! The problem is that the device doesnt wait for the WPS setup (which can take some time) to be completed. Therefore I simply put a while loop that checks if the setup was completed like so: (includes a timeout after 30s)
'''
WiFi.mode(WIFI_STA);
bool wpsSuccess = WiFi.beginWPSConfig();
if(wpsSuccess) {
  String newSSID = WiFi.SSID();
  if(newSSID.length() > 0) {
    DEBUG_WM(F("WPS Success"));
    unsigned long start = millis();
    while (WiFi.status() != WL_CONNECTED && millis() - start < 30000) {
      yield();
    }
    if (WiFi.status() == WL_CONNECTED) {
      DEBUG_WM(F("WPS Connected"));
      connect = true;
      ESP.reset();
    } else {
      DEBUG_WM(F("WPS Failed"));
      connect = false;
    }
  } else {
    DEBUG_WM(F("WPS Failed"));
    connect = false;
  }
}
'''
Just put this code instead of:
'''
WiFi.mode (WIFI_STA);
delay (1000);
startWPS ();
delay (5000);
ESP.reset ();
delay (2000);
'''
and it should work (at least on my machine)

tablatronix, 2023-01-22:
Does anyone still use WPS, isnt it a vulberability