tzapu/WiFiManager #368: wifiManager.autoConnect SSID not present on wifi

josep112, 2017-05-10:
I have a serious problem in my esp8266 esp12, they show me the AP in the serial with the AutoConnectAP name but the same does not appear on the wifi of the phone or the computer, I am using the wifimanager version 0.12, any help is welcome

*WM: AutoConnect
bcn 0
del if1
mode : sta(a0:20:a6:16:27:75)
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
mode : sta(a0:20:a6:16:27:75) + softAP(a2:20:a6:16:27:75)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

closed 2018-01-26 by tablatronix

flyingstarlai, 2018-05-10:
how to solve this issue?

tolgahansusur, 2019-12-16:
Just erase flash of esp 8266 totally before uploading code. It will work.

https://www.espressif.com/en/support/download/other-tools?keys=&field_type_tid%5B%5D=14
You can download eraser app from link above.