tzapu/WiFiManager #370: Change default SoftAP SSID & password on runtime from config page.

bkrajendra, 2017-05-23:
Just like routers, I want to have feature to change default AP mode SSID and/or password.
As WiFiManager currently goes into AP mode if its not able to connect to saved Wlan ssid/key, it can be little security issue. Any one who knows default AP password can connect and modify WiFi config.

So I want it like professional routers, where they provide default SSID n password in manual and then they offer it to change when login for first time. Also it can be changed latter on.

bkrajendra, 2018-01-26:
Dear @tablatronix  anything wrong with post?

tablatronix, 2018-01-26:
There is no default ap password you specify it in your code
`autoconnect(SSID,PASSWORD);`

bkrajendra, 2018-01-26:
What I mean by default password is the one we provide in code which is hardcoded in autoconnect(SSID,PASSWORD);. We can not change this at runtime after flashing hardware.

What i mean was, is it possible to create an interface for changing this AP password, like we do in routers to change admin password.?

tablatronix, 2018-01-26:
Yeah it can be done with parameters by user, not sure it needs to be in core. Maybe someone can make an example

bkrajendra, 2018-01-27:
ok.. I'll give it a try.

mrlightsman, 2018-03-07:
@bkrajendra 
Did you ever figure out a way to use parameters to change the ESP SSID and PW via the captive portal?  If so, please share how you did it. With code example, preferably.  Thanks.

bkrajendra, 2018-03-07:
@tablatronix 
I guess its kind of tricky and Ive not tried yet but can be done. To start of, just use parameter to store ssid and password, then just retrieve them back while setting up.
```
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

char ap_ssid[18] = "admin";
char ap_pass[18] = "admin";
//flag for saving data
bool shouldSaveConfig = false;
//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup()
{
  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");
          strcpy(ap_ssid, json["ap_ssid"]);
          strcpy(ap_pass, json["ap_pass"]);
        } else {
          Serial.println("failed to load json config");
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }

  WiFiManagerParameter custom_ap_ssid("SSID", "ap ssid", ap_ssid, 18);
  WiFiManagerParameter custom_mqtt_port("PASS", "ap passport", ap_pass, 18);

 //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

 //add all your parameters here
  wifiManager.addParameter(&ap_ssid);
  wifiManager.addParameter(&ap_pass);

 if (!wifiManager.autoConnect(ap_ssid, ap_pass)) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }
Serial.println("connected...yeey :)");
  strcpy(ap_ssid, custom_ap_ssid.getValue());
  strcpy(ap_pass, custom_ap_pass.getValue());

 //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    json["ap_ssid"] = ap_ssid;
    json["ap_pass"] = ap_pass;
    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());

}
void loop() {
  // put your main code here, to run repeatedly:

}
```
It can be like this.
Not tested.