tzapu/WiFiManager #372: How to open a web page when connect successs?

tarzan115, 2017-06-02:
hi @tzapu 
thank you for your library.

when I connect success to a wifi I want to open the web with notification "Success" after window auto close. or "Faile" when connecting unsuccess. how to do that?

thank you so much

tzapu, 2017-06-02:
hi,

Thanks for the appreciation.

Unfortunately you can’t really do that, as by that point you will have most likely been disconnected from the module.

cheers

tarzan115, 2017-06-02:
I tried to modified function `handleWifiSave` by connecting wifi inside with `connect_status` after that check connection before sending web page to the server. when connecting success it showed "Success" on the web page before the auto close window. but when connecting failed, it can't show "Failed" in the web page. maybe you can figure out that situation.
```c
void WiFiManager::handleWifiSave() {
  DEBUG_WM(F("WiFi save"));

  //SAVE/connect here
  _ssid = server->arg("s").c_str();
  _pass = server->arg("p").c_str();

  int connect_status = connectWifi(_ssid, _pass);
  //parameters
  for (int i = 0; i < _paramsCount; i++) {
    if (_params[i] == NULL) {
      break;
    }
    //read parameter
    String value = server->arg(_params[i]->getID()).c_str();
    //store it in array
    value.toCharArray(_params[i]->_value, _params[i]->_length);
    DEBUG_WM(F("Parameter"));
    DEBUG_WM(_params[i]->getID());
    DEBUG_WM(value);
  }

  String page = FPSTR(HTTP_HEAD);
  page.replace("{v}", "Credentials Saved");
  page += FPSTR(HTTP_SCRIPT);
  page += FPSTR(HTTP_STYLE);
  page += _customHeadElement;
  page += FPSTR(HTTP_HEAD_END);
  // page += FPSTR(HTTP_SAVED);
  page += "<h1>";
  if(connect_status == WL_CONNECTED)
  {
      page += "Success";
  }
  else
  {
      page += "Failed";
  }
  page += "</h1>"
  page += FPSTR(HTTP_END);

  server->sendHeader("Content-Length", String(page.length()));
  server->send(200, "text/html", page);

  DEBUG_WM(F("Sent wifi save page"));

  connect = true; //signal ready to connect/reset
}
```

tablatronix, 2017-08-27:
This is hard to handle but it can be done, the problem is you have to disconnect ap to negotiate sta, and if sta fails then sta+ap is not even possible. So you have to disconnect sta and restart the ap all while having the page refresh properly.

It is definetly on list

closed 2017-08-28 by tarzan115