tzapu/WiFiManager #373: How can i add another button to template.html

erdemontas, 2017-06-02:
I'm trying to add another form to your template but I'm a newbie in html So I felt like to ask.
Here is the question; How can i add another button and get two values using two form.Main purpose is getting min-max values from user and activate a buzzer for a weather station project.
First of all; which file should I edit to manipulate template. template.h or WiFiManager.template.html

Secondly I added  3rd line and I wanna get two integer as temp values. But esp12 still works with old template.
`<form action=\"/wifi\" method=\"get\"><button>Configure WiFi</button></form><br/>
          	<form action=\"/0wifi\" method=\"get\"><button>Configure WiFi (No Scan)</button></form><br/>
         	 <form action=\"/setTemp\" method=\"get\"><button>Set Temp Values</button></form><br/>
         	<form action=\"/i\" method=\"get\"><button>Info</button></form><br/>
         	<form action=\"/r\" method=\"post\"><button>Reset</button></form>`

tzapu, 2017-06-02:
hi

look into documentation for adding custom parameters

good luck

erdemontas, 2017-06-02:
I added
`WiFiManagerParameter custom_text2("<form action=\"/setTemp\" method=\"get\"><button>Set Temp Values</button></form><br/>"); `
into my code but when I clicked to newly added button (created in the Configure Wifi page) it directs me to [192.168.4.1/wifisave?s=&p=](url)

erdemontas, 2017-06-03:
It addes to Configure Wifi page when I add custom parameter, is there any way to add somewhere else?

brendanvanbreda, 2017-06-08:
@erdemontas where exactly do you want it added?

erdemontas, 2017-06-08:
@brendanvanbreda to main screen. When you first enter to 192.168.4.1 you see 4 button
Configure Wifi
Configure Wifi(no scan)
Info
Reset

I want to have another button in this page. I also tried to change template.h and wifimanager.cpp in the library folder but It still shows the same screen.
PS: I successfully added two parameter with using custom parameter functions but thats not exactly what I want

brendanvanbreda, 2017-06-08:
@erdemontas have a look at the functions `WiFiManager::handleWifiSave()` and  `WiFiManager::handleWifiSave()`. If you understand how the ESP8266 generates and serves the web pages. You need to create something similar and include these inside the `WiFiManager::setupConfigPortal()` in the setup web pages sections in this function inside the WifiManager.cpp. You can move the parameter generation part from the `WiFiManager::handleWifi()` to your new function and you should be able to view this as a new button once you edit the WifiManager.h `HTTP_PORTAL_OPTIONS[]` to add a new button. I have this working on my own implementation.

Quite a bit of extra code needs to be added in order to be able to do this but it can be done.

Good Luck

oscarmoutinho, 2017-07-26:
I wanted to use only the wifi config page without modifying the core, so I added:
wifiManager.setCustomHeadElement("<script>if (window.location.href.indexOf(\"/wifi\")<0) {window.location=\"/wifi\"};</script>");
So the wifi config page is shown (/wifi[?...]) and save page (/wifisave) is not blocked.
You may explore this approach.

tablatronix, 2017-08-27:
Hooks will be added if they do not already exist

closed 2018-01-26 by tablatronix