tzapu/WiFiManager #379: Non blocking AP mode

git-tiger, 2017-06-28:
I have a project that requires the ESP8266 to startup in access point mode, but the remaining code is blocked till the access point is configured.
Is there any way to continue with the code, and still have access point available in the background?

synfinatic, 2017-06-30:
I've got the same requirement and from what I can tell, there isn't a way to do this in the current code today.

ANASOMARY, 2018-03-27:
Hi Guys,

Is this feature added? am really in bad need for this.

tablatronix, 2018-03-27:
development
    //if this is set, portal will be blocking and wait until save or exit, is false user must manually `process()` to handle config portal
    void          setConfigPortalBlocking(boolean shouldBlock);

ANASOMARY, 2018-03-27:
Hi, 
Really appreciate your reply.
I did not get the process() part , but if I wanted the code to continue to the loop even after going into AP mode, then should I set setConfigPortalBlocking(false); ??

tablatronix, 2018-03-27:
yes and process() in your loop

ANASOMARY, 2018-03-27:
void loop()
{
   _**process();**_
  HandleSwitch();
  server.handleClient();
  delay(10);
}

Thats it??? Oh god thanks Shawn, you really saved my @$$. 💯

tablatronix, 2018-03-27:
Yeah I mean its development, and hardly tested...

ANASOMARY, 2018-03-27:
Oh snap
[image]

tablatronix, 2018-03-27:
https://github.com/tzapu/WiFiManager/issues/500

ANASOMARY, 2018-03-27:
Thanks Shawn,

does it support generic esp8266? am using a sonoff, and am getting the following when I compile:
[image]

tablatronix, 2018-03-27:
It should

ANASOMARY, 2018-03-27:
Sonoff Basic

ANASOMARY, 2018-03-27:
esp8266

tablatronix, 2018-03-27:
esp8266 which one?

ANASOMARY, 2018-03-27:
ESP8266 V4.1

ANASOMARY, 2018-03-27:
Hi Shawn,
my mistake, instead of Process() I should have put  wifiManager.process(); same for setConfigPortalBlocking
It is compiling with no errors now.
Thanks again.

ANASOMARY, 2018-03-27:
Am Sorry Shawn, But I was using the wrong library, am still getting the same compilation error for generic esp8266
[image]

ANASOMARY, 2018-03-27:
This is strange... I tried compiling for a different board, and am getting the same error!
[image]

tablatronix, 2018-03-27:
You are gonna have to turn on verbose compilation warnings or whatever adruino has to show some actual info

tablatronix, 2018-03-27:
https://github.com/tzapu/WiFiManager/issues/524

hassasian, 2018-04-19:
@ANASOMARY you should update your board software from board manager. You should use 2.4.1 or newer one.

hassasian, 2018-04-19:
@tablatronix thx for ur help. But I have one more question. Our esp connected to an router. After that I close the router. Then as suppose our esp come back to AP mode. When I connected ap of esp, it cannot handle http requests. Sorry for bad eng.

tablatronix, 2018-04-19:
not sure i understand

hassasian, 2018-04-19:
Give me one more shot to explain it :) I have a esp8266 module. I'm using wifimanager development branch for nonblocking ap mode. its working great. Congrats. My problem is that ; when esp cannot access the router which it already connected it come back to access mode.  And I connect the AutoConnectAP with my computer to configure my esp for another router. But esp cannot answer  my http request.

hassasian, 2018-04-19:
in addition if you reset your router which esp connected. Esp disconnected from router and enabling its ap. After your router started. Esp connect the router succesfully but esp dont close its ap. Briefly if you restart your router, your esp will enable its ap mode. After your router come back esp will connect it but dont close its ap.

tablatronix, 2018-04-19:
The esp should reconnect by itself

tablatronix, 2018-05-12:
I do not think this is handled automatically, since when in configportal usually STA mode is disabled so AP mode works properly.

It is not wise, to do so , and has been updated in hotfixes branch and development as such. If STA is failing it makes AP mode very unstable.

But if you wanted to, in development branch , (if you hardcode channel for stability and leave sta on), you can call `stopConfigPortal` at any time.

Kirk-1, 2018-06-15:
Hi @tablatronix, Thanks for all your work on this. 
I had AutoConnect working in non-blocking mode beautifully for a while. Now I am having an issue with:
W (84) wifi: wifi nvs_open fail ret=4353
E (84) wifi: wifi_init 1410 ret=4353

So now it seems it wont save the ssid and password once entered in the portal. I am wondering if you could shed any light on this. Perhaps nvs_init() needs to be called somewhere..?

Any help greatly appreciated!

Cheers,

Kirkland

tablatronix, 2018-06-15:
Is this an esp32 or eps8266?

Kirk-1, 2018-06-15:
ESP32 Wrover

tablatronix, 2018-06-15:
Try erasing nvs, there was a bug in esp32 lib a while ago.
You can find instructions in esp32 issues.

Kirk-1, 2018-06-15:
Tried using pytool to erase. Tried using nvs_erase in sketch. Tried nvs_init at start of setup. Tried updating esp libraries. Tried the specific Webserver library recommended.
I use wifimanager.process() until I have a connection, then I swap over to serving a web page from within the sketch. So I have an instance of WiFi client, mdns and webserver in the sketch also. Perhaps this is the source of the problem. It was working fine but now this nvs issue. I will keep hunting... Thankyou. Cheers, Kirk

Kirk-1, 2018-06-18:
Thanks again. It seems the problem occurs when I declare an instance of WiFiManager globally. I need to use wifiManager.process() in my loop and so it wont compile unless I declare wifiManager in the loop. But declaring and redeclaring wifiManager continuously in the loop isn't going to work either. I noticed that the same error occurs in the "OnDemandConfigPortal" example because, it would seem, the WiFiManager is declared globally. I am using the development version of WIFiManager.

Kirk-1, 2018-06-18:
Can I enable debug compile flags easily using Arduino?

tablatronix, 2018-06-18:
I think there is a menu in board settings debug level

tablatronix, 2018-06-18:
ok I think I know what the problem might be.
I assume you are using the latest esp32 also ?

tablatronix, 2018-06-18:
esp32 introduced some new bugs, I just fixed them ( or rather worked around them )
Give it a try now.

The wifimanager constructor calls
getmode (causing the nvs_open failure error )
and persistent(false) which when called before esp32 calls wifiLowLevelInit(_persistent) now turns off flash storage of persistent, even if you change it later.

Kirk-1, 2018-06-19:
That did it! Thank you very much!

xserg, 2018-06-28:
@tablatronix Hi! Can you pls add some example for non- blocking case?

tablatronix, 2018-06-28:
You can check the ondemandwebportal example

xserg, 2018-06-28:
Have an issue with wifisave - it crashes esp on develop branch, on master it ok. tested on Nodemcu esp12 and D1 R1 mini
wifisave page time outed, credentials not saved

tablatronix, 2018-06-28:
what esp version?

xserg, 2018-06-28:
esp 2.4.1, 
esp 12f nodemcu
autoconnect example and my scetch works the same, cant save setting and cant open wifisave from info page
Thanks

tablatronix, 2018-06-28:
do you get a exception or dump?

xserg, 2018-06-29:
*WM: <- HTTP WiFi save  
*WM: Parameters 
*WM: -------------------- 
*WM: server: mqtt.********
*WM: port: 1883
*WM: username: **********
*WM: password: ***********
*WM: -------------------- 

Exception (3):
epc1=0x40217eb8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402521a2 depc=0x00000000
ctx: cont 
sp: 3fff0860 end: 3fff0e10 offset: 01a0
>>>stack>>>
3fff0a00:  3ffefcf4 000005d0 3fff0a4c 402144b1  
3fff0a10:  3fff2704 00000000 3fff0a4c 402144fe  
3fff0a20:  402521a2 00000000 3fff0c30 402143d8  
3fff0a30:  3fff2704 00000000 3fff0c30 40211ba9  
3fff0a40:  00000000 00000000 00000000 3fff41bc  
3fff0a50:  0000001f 0000001b 3fff420c 00000b0f  
3fff0a60:  00000b06 00000009 3fff0ab0 4021447f  
3fff0a70:  3ffefcf4 000005d0 000005d0 4020c0fc  
3fff0a80:  00000002 00000001 3fff2ab4 40216337  
3fff0a90:  00000000 00000000 3fff2ab4 4020c0f2  
3fff0aa0:  3fff2ab4 3fff2748 3fff2ab4 4020c12e  
3fff0ab0:  00000000 00000000 00000000 402145dc  
3fff0ac0:  3fff2ab4 3fff2748 3fff2704 4020c1b5  
3fff0ad0:  3fff3914 0000000f 00000009 00000000  
3fff0ae0:  00000000 00000000 3fff2708 4020a424  
3fff0af0:  00000057 00000001 00000001 00000001  
3fff0b00:  00000002 3fff2f64 00000057 3fff2ab4  
3fff0b10:  00000000 3fff31a4 3fff2704 3ffe8d89  
3fff0b20:  00000001 3fff272c 3fff2704 4020c404  
3fff0b30:  00000000 00000001 3fff23fc 4020b3ad  
3fff0b40:  00000000 3fff31a4 3fff0c30 402143d8  
3fff0b50:  00000000 00000001 3fff0c30 4021200b  
3fff0b60:  00000000 00000000 3ffefdf0 00000001  
3fff0b70:  00000000 3ffe8d89 40214f7c 3ffefdf0  
3fff0b80:  00000000 00000001 3fff0c30 40214ff4  
3fff0b90:  00000000 00000001 3fff0c30 40212b1d  
3fff0ba0:  4025208e 00000199 00000199 4010020c  
3fff0bb0:  3ffe8c03 3ffefad0 3fff0bec 3ffe8d89  
3fff0bc0:  3ffe8d89 3ffefad0 3fff0c30 4020fe96  
3fff0bd0:  3ffe8c03 3ffefad0 3fff0c30 40212cd0  
3fff0be0:  00000000 00000000 00000000 00000000  
3fff0bf0:  00000000 00000000 3fff0c30 4020f514  
3fff0c00:  00000000 00000001 0000005f 40212d29  
3fff0c10:  feefeffe feefeffe 00000000 3ffef929  
3fff0c20:  3ffe8544 3ffefd1c 3fff0d30 40208da0  
3fff0c30:  3fff23fc 3fff2704 3fff1fb4 3fff1fb7  
3fff0c40:  3fff1fb8 3fff1f2c 3fff1f38 3fff1f38  
3fff0c50:  40107338 00000000 40107338 00000000  
3fff0c60:  40107338 00000000 40107338 00000000  
3fff0c70:  40107338 00000000 40107338 00000000  
3fff0c80:  40107338 00000000 feef5035 3fff1f3c  
3fff0c90:  0000000f 00000004 3fff1f54 0000000f  
3fff0ca0:  00000000 3fff1f6c 0000000f 00000002  
3fff0cb0:  3fff1f84 0000000f 0000000a 00000000  
3fff0cc0:  00000000 00000000 00008210 0000821a  
3fff0cd0:  00000001 3fff1f9c 0000000f 00000003  
3fff0ce0:  feefef00 00000001 000042d5 000007d0  
3fff0cf0:  feefef07 ffffffff 00000000 00000000  
3fff0d00:  01000001 00010101 01010001 feefef01  
3fff0d10:  3ffe8c03 3ffe8c03 00000001 01000000  
3fff0d20:  01000000 00007531 00000004 00000005  
3fff0d30:  3fff1fbc feef0201 3ffefd1c feefeffe  
3fff0d40:  feefeffe 00000000 feefeffe 40207020  
3fff0d50:  3ffefd1c 40215f70 40215f5c 3ffe8c51  
3fff0d60:  3ffe8c51 3fff1f04 00000020 00000001  
3fff0d70:  3ffe8c03 3ffe8c48 3ffe8c48 3fff1edc  
3fff0d80:  00000020 00000001 3ffe8c03 3ffe8d3b  
3fff0d90:  3ffe8d36 3fff1ecc 00000005 00000001  
3fff0da0:  3ffe8c03 3ffe8d2f 3ffe8d2a 3fff1e9c  
3fff0db0:  00000028 00000001 3ffe8c03 40207020  
3fff0dc0:  feefeffe 40215f70 40215f5c feefeffe  
3fff0dd0:  00000000 feefeffe feefeffe feefeffe  
3fff0de0:  feefeffe feefeffe feefeffe 3ffefddc  
3fff0df0:  3fffdad0 00000000 3ffefdd4 40214fc0  
3fff0e00:  feefeffe feefeffe 3ffefdf0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

vaz83, 2018-07-13:
Hi, i have a nodeMCU board, parameters work fine without setting the wifiManager no nonblocking. Once i set it to true, it gives exception (3) as resets. Seems that when adding custom parameters is not compatible with nonblocking option. Are you guys fixing this? thanks

tablatronix, 2018-07-14:
Cant fix if I cant reproduce.
Make a new issue with small example.

tablatronix, 2018-07-14:
I will test some more

tablatronix, 2018-07-14:
I added some overflow fixes try now

vaz83, 2018-07-14:
Will try now and let you know.

vaz83, 2018-07-14:
Hi tablatronix, i've just tested, not working, now, even the parameters are not showing up, this is my code:

`#include <FS.h>
#include <ESP8266WiFi.h> //ESP8266 Core WiFi Library (you most likely already have this in your sketch)
#include <DNSServer.h> //Local DNS Server used for redirecting all requests to the configuration portal
#include <ESP8266WebServer.h> //Local WebServer used to serve the configuration portal
#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <ArduinoJson.h>

WiFiManager wifiManager;
WiFiEventHandler gotIpEventHandler, disconnectedEventHandler;
char mqtt_server[40] = "0.0.0.0";
char mqtt_topicIN[80] ="IOT_Device/IN";
char mqtt_topicOUT[80] ="IOT_Device/OUT";
char mqtt_port[6]  = "1883";

/*
void handleRoot() {
  wifiManager.startWebPortal();
}

void handleReset() {
  server.send(200, "text/plain", "Resetting!");
    //Serial.println("diconnecting client and wifi");
    //client.disconnect();
    wifi_station_disconnect();

  WiFiManager wifiManager;
  wifiManager.resetSettings();
  ESP.reset();
  
}

void handleNotFound(){
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}
*/

void setup()
{
  Serial.begin(115200);
  Serial.println("Setup mode...");
  //wifiManager.resetSettings();
   gotIpEventHandler = WiFi.onStationModeGotIP([](const WiFiEventStationModeGotIP & event)
  {
    Serial.print("Station connected, IP: ");
    Serial.println(WiFi.localIP());
  });

   disconnectedEventHandler = WiFi.onStationModeDisconnected([](const WiFiEventStationModeDisconnected & event)
  {
    Serial.println("Station disconnected!");
  });
  WiFiManagerParameter custom_text1("<p><b>MQTT Settings</b></p>");
  WiFiManagerParameter custom_mqtt_server("server", "Server IP", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "Port", mqtt_port, 6);
  WiFiManagerParameter custom_mqtt_topicIN("topicIn", "Input Topic", mqtt_topicIN, 80);
  WiFiManagerParameter custom_mqtt_topicOUT("topicOut", "Output Topic", mqtt_topicOUT, 80);
  wifiManager.addParameter(&custom_text1);
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_mqtt_topicIN);
  wifiManager.addParameter(&custom_mqtt_topicOUT);

  wifiManager.setConfigPortalBlocking(false);
  
  wifiManager.autoConnect("IOT Device");

  
  
  /*
  server.on("/", handleRoot);

  server.on("/inline", []()
  {
    server.send(200, "text/plain", "this works as well");
  });
  server.on("/resetwifi", handleReset);
  server.onNotFound(handleNotFound);
  server.begin();
  */
  Serial.println("HTTP server started");
}

void loop()
{
  wifiManager.process();
 
  Serial.println("hello world");
  // put your main code here, to run repeatedly:
}`

Now, even if you take the setConfigPortalBlocking, the parameters are no longer showing up.

In the serial I got this:

*WM: <- Request redirected to captive portal 
*WM: <- HTTP Root 
*WM: Scan is cached 
*WM: <- HTTP Wifi 
*WM: Scan is cached 
*WM: 1 networks found
*WM: AP:  -35 V@Z Wifi
*WM: [ERROR] WiFiManagerParameter is out of scope 
*WM: <- Request redirected to captive portal 
*WM: <- HTTP Root 
*WM: Scan is cached

vaz83, 2018-07-14:
Hi Tablatronix,
I have done what you said (put the variabes global at the beggining), the error still persists. Like I said, even if I comment the setConfigPortalBlocking(false), no parameters are shown and error *WM: [ERROR] WiFiManagerParameter is out of scope  is shown. with the version of WM i had from yesterday it was working fine.

tablatronix, 2018-07-14:
works for me

```CPP
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

WiFiManager wm;
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", "servername", 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", "1234", 6);

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
    delay(3000);

    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around

    // add a custom input field

  WiFiManagerParameter custom_field("customfield", "Custom Field", "Default Value", 40);
  WiFiManagerParameter custom_token("api_token", "api token", "0123456789", 9); // overflow
  // WiFiManagerParameter custom_tokenb("invalid token", "invalid token", "abcdefghijk", 11);

  //add all your parameters here
  wm.addParameter(&custom_mqtt_server);
  wm.addParameter(&custom_mqtt_port);
  wm.addParameter(&custom_field);
  wm.addParameter(&custom_token);
  // wm.addParameter(&custom_tokenb); // overflow

    //reset settings - wipe credentials for testing
    //wm.resetSettings();

    wm.setConfigPortalBlocking(false);
    if(wm.autoConnect("AutoConnectAP")){
        //if you get here you have connected to the WiFi
        Serial.println("connected...yeey :)");
    }
    else 
    {
        Serial.println("non blocking config portal running");
    }

    wm.startConfigPortal("OnDemandAP");
}

void loop() {
    wm.process();
    // put your main code here, to run repeatedly:
    
}

```

vaz83, 2018-07-14:
this is giving *WM: [ERROR] WiFiManagerParameter is out of scope .. Am I doing something wrong?

`#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

WiFiManager wm;
char mqtt_server[40] = "0.0.0.0";
char mqtt_topicIN[80] ="IOT_Device/IN";
char mqtt_topicOUT[80] ="IOT_Device/OUT";
char mqtt_port[6]  = "1883";

WiFiManagerParameter custom_mqtt_server("server", "Server IP", mqtt_server, 40);
WiFiManagerParameter custom_mqtt_port("port", "Port", mqtt_port, 6);
WiFiManagerParameter custom_mqtt_topicIN("topicIn", "Input Topic", mqtt_topicIN, 80);
WiFiManagerParameter custom_mqtt_topicOUT("topicOut", "Output Topic", mqtt_topicOUT, 80);

void setup()
{
  Serial.begin(115200);
  delay(3000);
  Serial.println("Setup mode...");
  //wifiManager.resetSettings();
 
  wm.addParameter(&custom_mqtt_server);
  wm.addParameter(&custom_mqtt_port);
  wm.addParameter(&custom_mqtt_topicIN);
  wm.addParameter(&custom_mqtt_topicOUT);

  wm.setConfigPortalBlocking(false);

  if(wm.autoConnect("IOT_Device"))
  {
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
  }
  else 
  {
    Serial.println("non blocking config portal running");
  }
  wm.startConfigPortal("IOT_Device"); 
  
}

void loop()
{
  wm.process();
 
  //Serial.println("hello world");
  // put your main code here, to run repeatedly:
}`

tablatronix, 2018-07-14:
testing

tablatronix, 2018-07-14:
I dont see it

[image]

tablatronix, 2018-07-14:
I am on esp32, should I test on 8266 also ?

vaz83, 2018-07-14:
Can you please try on the 8266? Is the one I am using

tablatronix, 2018-07-15:
Will do

vaz83, 2018-07-15:
Hi tablatronix, were you able to test it on the esp8266 ? Parameters are not showing up :(

[11 more replies]