tzapu/WiFiManager #381: Error calling configModeCallback inside another class

jcmojj, 2017-07-01:
I made the class Internet to take out of the main code all the code from wifiManager and make it cleaner, but I have a problem with the configModeCallback that have an error that I don't know how to fix.

The internet class is:
```
Internet::Internet(bool resetWifi){
  Serial.begin(115200);
  wifiAutoConnect(resetWifi);
  pinMode(TRIGGER_PIN, INPUT);
}
void Internet::wifiAutoConnect(bool resetWifi){ 
    WiFiManager wifiManager;
    wifiManager.setBreakAfterConfig(true);
    wifiManager.setAPCallback(configModeCallback);
    if(resetWifi){wifiManager.resetSettings();}//reset saved settings

      wifiManager.setMinimumSignalQuality(20);
      wifiManager.setTimeout(120); 

    if (!wifiManager.autoConnect("MyNet","12345678")) {
      Serial.println("failed to connect and hit timeout");
      ESP.restart();
      delay(1000);
    }
    Serial.println("connected...yeey :)");
}
void Internet::wifiCheckToConnectOnDemand(bool resetWifi){ 
  if ( digitalRead(TRIGGER_PIN) == LOW ) {
      WiFiManager wifiManager;
      if(resetWifi){wifiManager.resetSettings();}//reset saved settings
      wifiManager.setMinimumSignalQuality(20);
      wifiManager.setTimeout(120); 
      wifiManager.setBreakAfterConfig(false);
      if (!wifiManager.startConfigPortal("MyNet","12345678")) {
        Serial.println("failed to connect and hit timeout");
        delay(3000);
        //reset and try again, or maybe put it to deep sleep
        ESP.restart();
        delay(5000);
      }
    Serial.println("connected...yeey :)");
  }
}
void Internet::configModeCallback(WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode(callBack):");
  Serial.println(WiFi.softAPIP());
}
```
I call from setup: with `internet.wifiAutoConnect(resetWifi);`

I call from loop  with `internet.wifiCheckToConnectOnDemand(resetWifi);`

I get this error message:

> Arduino: 1.8.3 (Mac OS X), Placa:"ESP32 Dev Module, 80MHz, 921600, None"
> Internet.ccp.ino: In member function 'void Internet::wifiAutoConnect(bool)':
> Internet.ccp:21: error: invalid use of non-static member function
>      wifiManager.setAPCallback(configModeCallback);

 I know that the code expected that the configModeCallback was inside of the main code and not inside another class. If someone can give me a tip to change something in my code or inside the lib, you are welcome. To make it easy the part of the lib that deals with this function is:

```
void WiFiManager::setAPCallback( void (*func)(WiFiManager* myWiFiManager) ) {
  _apcallback = func;
}
```

tablatronix, 2017-08-28:
You probably need to put wifimanager globally across your class, you are calling it in each function.

suculent, 2017-09-06:
I have similar issue.

``` language:cpp
// .h
class MyClass {
   public:
       WiFiManager *manager;
       void saveConfigCallback();
 }

// .cpp
class MyClass {

    void some_initializer() {
          manager = new WiFiManager();
          manager->setSaveConfigCallback( saveConfigCallback );
    }

    void THiNX::saveConfigCallback() {
      should_save_config = true;
    }
}
```

And the error. It probably needs just some casting. One issue is that when dropping WiFiManager into a C++ class, you need to change dot notation from examples to "->" and even more:

```
src/MyClass.cpp:5:4: error: no matching function for call to 'WiFiMana
ger::setSaveConfigCallback(<unresolved overloaded function type>)'
manager->setSaveConfigCallback( saveConfigCallback );
^
src/MyClass.cpp:5:4: note: candidate is:
In file included from src/MyClass.h:8:0,
from src/MyClass.cpp:1:
lib/WiFiManager/WiFiManager.h:101:19: note: void WiFiManager::setSaveConfigCallback(void (*)())
void          setSaveConfigCallback( void (*func)(void) );
^
lib/WiFiManager/WiFiManager.h:101:19: note:   no known conversion for argument 1 from '<unresolved over
loaded function type>' to 'void (*)()'

```

suculent, 2017-09-06:
Here is the whole source https://github.com/suculent/thinx-firmware-esp8266

suculent, 2017-09-06:
Which just reminds me how I have solved that recently. So to @jcmojj 's issue:
a) Move your WiFiManager *manager; line to class header and initialize with manager = new WiFiManager();
b) I needed also to set the saveConfigCallback and compiler did complain about <unitialized overloaded function something...>  so I have inlined the callback into the header like so:

```
class MyClass {

  public:
 
    WiFiManager *manager;
    WiFiManagerParameter *api_key_param;

    // when user sets new API Key in AP mode
    void saveConfigCallback() {
      should_save_config = true;
      strcpy(thx_api_key, api_key_param->getValue());
    }
}
```

This compiles and links without errors.

suculent, 2017-10-30:
Thanks to this example: https://github.com/ajfisher/espixel/blob/master/firmware/esp8266_pixel/configurator.cpp

Solved by turning callback (and everything accessed from the callbac) to static. This resolved the `<unresolved overloaded function type>` error.

tablatronix, 2017-10-30:
oh ok, maybe post example code for others with the same issue.

closed 2018-01-26 by tablatronix

keitetran, 2018-07-04:
add static to var

Look https://github.com/ajfisher/espixel/blob/master/firmware/esp8266_pixel/configurator.h#L52