tzapu/WiFiManager #391: Open config portal from link

StoyanVP, 2017-07-25:
Hello. 
Thanks to tzapu and kentaylor for WiFiManager librarie.
I have a question. 
Is there a suitable method to call the configuration portal from link in the web page?
Once the settings have already been made, then you can only click on link and open portal to change the network settings without pressing a hardware button for that.  Then go back to the web page.

ageurtse, 2018-02-16:
this is also what i'm looking for !

tablatronix, 2018-02-16:
you can leave the webserver open permanently and not use ap mode at all, is that what you are asking ?

ageurtse, 2018-02-16:
i think.

what i wan't first setup goto failover en set ssid, password and if needed ip settings, when ipsetting not provided dhcp mode should be enabled. 

after setting up for the first time, the esp reboots and starts using those setings.

when this is done, i wan't the user give the ability to enter the config page again to alter stuff. 
i'm also running a webserver to control stuff. 

is this possible.

tablatronix, 2018-02-16:
I think the webserver will conflict with wifimanager, we have yet to figure out the best way to share the webserver, you can make it public and use it instead in code, or we can pass it in to wifimanager to use or some other c++ way to shared ptr or what not.

sidey79, 2018-03-29:
Is there an option, to prevent generating the own AP on the esp  if there is a connection when starting via 
startWebconfigPortal() and running process() in loop?

tablatronix, 2018-03-30:
No but you can start the webserver the same way in development branch. See the example