tzapu/WiFiManager #397: Add Custom Parameters

nimaaryamehr, 2017-08-13:
In this program, I want to get the TOKEN_NOTIFY parameter from the user on the configuration page
How to do it
I also read the parameter "Custom Parameters" but I did not notice anything

`#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

const byte interruptPin = 2;

const char* BASEURI="http://telegramiotbot.com/api/notify?token=";
const char* TOKEN_NOTIFY="597ec780968fb";

bool interrupted=false;
/*
    interrupt handler
*/
void inputChanged(){
    interrupted=true;
}
void notify(const char* token){
    HTTPClient http;

    http.begin(String(BASEURI) + token);

    int httpCode = http.GET();

}

void setup() {
    
    WiFiManager wifiManager;
    //wifiManager.resetSettings();
    wifiManager.autoConnect("SmartHome");
    
    pinMode(interruptPin, INPUT_PULLUP);
    attachInterrupt(digitalPinToInterrupt(interruptPin), inputChanged, RISING);
}

void loop() {
    if(interrupted){
        if((WiFi.status() == WL_CONNECTED)) {
            notify(TOKEN_NOTIFY);
            interrupted=false;
        }
    }
}`

herostrat, 2017-09-19:
Hey @nimaaryamehr,

as far as I could understand you want the user to write a token in your configuration to be able to work with it in the code.

You can find an example of this int the [examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino](https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino)

Notice that you need to define custom parameters see e.g. [line 76](https://github.com/tzapu/WiFiManager/blob/89af31245b98b05e7d89319746ef3bc29d7e965b/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino#L76)

Than add this paramter to your instance of the wifimanager e.g. [line 91](https://github.com/tzapu/WiFiManager/blob/89af31245b98b05e7d89319746ef3bc29d7e965b/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino#L91)

And after the user put the information in the form and the connection to the network is done you can get the data via getValue() (e.g. [line 123](https://github.com/tzapu/WiFiManager/blob/89af31245b98b05e7d89319746ef3bc29d7e965b/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino#L123)) which copys the data in a char array.

I hope this helps.

Cheers

vaz83, 2018-07-13:
HI, i am also adding some parameters but i am noticing that the label of the field is now showing up, did this happened to anyone else =?

tablatronix, 2018-07-13:
development version adds labels.... is that what you mean?

vaz83, 2018-07-14:
Sorry tablatronix, I was using an old version of the WM, i downloaded the latest development version so this issue is solved already :) thanks :)

dontsovcmc, 2019-11-28:
@tablatronix close, solved

closed 2019-11-28 by tablatronix

Robotical2, 2020-12-31:
Hi  vaz83, can you please share which WM version you used for adding Custom Parameters.