tzapu/WiFiManager #401: Use custom settings fields ...

nardev, 2017-08-29:
It's quite a pain to extend WiFiManager now with custom fields..

I was playing a bit with that, so i added some code in WiFiManager, IMHO it's quite easier approach and much much much more versatile and adaptable to custom needs.

I didn't try to save space so i event added jquery and some more files. All compressed and separately server. (can be avoided)

You can  see all that in my project (under development, but this functionality works ok for demo)
https://github.com/nardev/hepek-iot

But regarding custom settings that i'm talking about, it's all in "lib/WiFiManagerCustom" link: https://github.com/nardev/hepek-iot/tree/master/lib/WiFiManagerCustom

Also, note, i added some options to: copy/compress files from "lib/WiFiManagerCustom/extras/..." to "data/". All that is still handled by same "parse.js" script, just little extended...

Essentially what i do is:

1. get .json file and "populate" already prepared form.  There is even loop for  forms which have same blocks like mqtt topics etc.

2. When saving, i parse form into a json and save it to a settings.json by just replacing the node which was submitted, all other "root" nodes are preserved.

So, it's "one form" == "one top level json node".

There should be one function checking consistency of posted data so that you can't POST empty node but some other checks are there.

I hope somebody will like this. (the project is not done. just this functionality)

tablatronix, 2017-08-29:
I have been looking into ultra micro js frameworks
Also I think a templating system would be nice.

Not sure about the best solution, obviously bloating the library or requiring internet access for cdns is not ideal.

Nice project,

nardev, 2017-08-30:
Well, i used jquery just to prove concept... it  takes about 30K on SPIFFS.. which is ok in my case... it's gzipped of course...

closed 2018-01-26 by tablatronix