tzapu/WiFiManager #402: resetsettings unreliable

tablatronix, 2017-08-29:
@tzapu I need to fix resetsettings not compatible with persistent false.
And there is no api to remove wifi credentials.

Do you know if there is a reason to avoid using ESP.erase_config ?

erase config seems to erase a chunk of memory, disconnect seems to erase config only station_config
which is ssid password bssid only

I guess erase will remove static ip etc.

Thoughts ? I would rather use a proper SDK function or write one to do as intended of this function.

tablatronix, 2017-08-29:
I have implemented some workarounds for now.
Although almost every device i have used requires a reboot after saving

tablatronix, 2017-08-29:
Not that i want to force a restart just leveraging the esp lib if possible

tarzan115, 2017-08-30:
```c
void WiFiManager::resetSettings() {
  DEBUG_WM(F("settings invalidated"));
  DEBUG_WM(F("THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA."));
  WiFi.persistent(true);
  WiFi.disconnect(true);
  WiFi.persistent(false);
}
```

try this maybe can help.

tablatronix, 2017-08-30:
my pr it is already using that

jwktje, 2018-01-28:
I'm confused about this too! After resetting the settings, it still autoconnects to the last network. Does setRestorePersistent(false) force this to not happen?

tablatronix, 2018-01-28:
I will have to test this, it should work in the development branch, let me know if you find otherwise and post logs

jwktje, 2018-01-28:
Logs show the settings being reset but it still connects. Would you like full logs?

tablatronix, 2018-01-28:
yes please

jwktje, 2018-01-28:
```
*WM: SETTINGS ERASED
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: wifi station disconnect
*WM: Connecting to saved AP
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: 
*WM: WL_CONNECTED
*WM: IP Address:
*WM: 192.168.2.25
Connected with IP address: 192.168.2.25
0:0
Server Started!
*WM: unloading
```

tablatronix, 2018-01-28:
is this an example ?

jwktje, 2018-01-28:
This is what the Logs show me after calling the WifiManager with this in setup:

```
  //Manager setup
  WiFiManager wifiManager;
  wifiManager.resetSettings();
  wifiManager.setBreakAfterConfig(true);
 
  if (!wifiManager.autoConnect("WifiSetup")) {
    Serial.println("failed to connect, we should reset as see if it connects");
    delay(3000);
    ESP.restart();
    delay(5000);
  }
  
  Serial.print ( "Connected with IP address: " );
  Serial.println ( WiFi.localIP() );
```

tablatronix, 2018-01-28:
try reseting after resetsettings please

jwktje, 2018-01-28:
But that would cause a loop if I called ESP.restart() in the Setup function. I basically just want a hardware reset to cause it to forget the SSID and password.

tablatronix, 2018-01-28:
Do it once to see if it fixes it

jwktje, 2018-01-28:
Tried it once by adding ESP.restart(); after the resetSettings();. 

As expected it started looping. And didn't fix the problem. 

Is there another way to force the ESP to forget it's last credentials in Setup?

tablatronix, 2018-01-28:
So after removing the reset it still autoconnects? Hmm

jwktje, 2018-01-28:
Yep exactly. Weird. Because I remember the resetSettings() working fine before I updated the library to the current ESP32 branch. Everything else is very stable now but this suddenly stopped working as expected. Any idea what could be the problem?

tablatronix, 2018-01-28:
Esp32 branch has not been tested against esp8266 fully either.

tablatronix, 2018-01-28:
Oh which esp version?

jwktje, 2018-01-28:
Wemos Lolin32 which has the ESP-WROOM-32. 
Running with this Library

```
{
  "name": "WifiManager",
  "keywords": "wifi, wi-fi",
  "description": "ESP8266 WiFi Connection manager with fallback web configuration portal",
  "repository":
  {
    "type": "git",
    "url": "https://github.com/tzapu/WiFiManager.git"
  },
  "frameworks": "arduino",
  "platforms": "espressif8266",
  "version": "0.12"
}

```

tablatronix, 2018-01-28:
Esp 2.4.0? Or staging?

tablatronix, 2018-01-28:
Oh and keep esp32 issues in the esp32 issue for now, this is specific to esp8266

tablatronix, 2018-01-29:
@jwktje try now

jwktje, 2018-01-29:
Works perfectly now! Great job, thanks! :D