tzapu/WiFiManager #403: WIFI_AP_STA mode after connecting

Dzavid, 2017-08-29:
Is it possible to switch ESP to dual mode (Access point and Station) after it's been successfully connected to WiFi network with DNS still working? I need this because my device will be given to the people who are not familiar with coding so I want them to be able to connect to ESP and go to config portal anytime, even if ESP is already connected. 

I guess I need to modify .cpp file and conditions in it, but it seems like DNS stops working after it switches to dual mode.

tablatronix, 2017-08-29:
What do you mean by dns?

Dzavid, 2017-08-29:
@tablatronix 

I'm sorry. I made a mistake. What I meant is that when I connect to ESP after it is connected to my WiFi, I can't reach configuration portal of WiFiManager. It seems like captive portal doesn't work anymore. It doesn't redirect my nor I can access it directly from IP (192.168.4.1).

tablatronix, 2017-08-29:
Hmm well the captive portal only really works on the softap, and that shuts down after you save.
Do you want it to stay on, via the softap, or do you want to get to it through the network, which would be harder.

papartmike, 2017-09-11:
I think I need something similar... I'd rather access the configuration portal also when connected as a client. What i'm doing so far is to create a web page as a web server that allows me to call the ondemand config portal (softAP). The only problem is that I need to have a web server on a different port (not 80) since probably something is going in a conflictuos state and config portal is not shown. Basically it should show the web server page also when connected as a client. Thanks

closed 2018-01-26 by tablatronix