tzapu/WiFiManager #409: Convert all form processing to POST

tablatronix, 2017-08-30:
Get is not a secure method of submitting forms on an open ap specifically.

- [ ] convert all to post
- [ ] support GET optionally ( for script based url programming )

tzapu, 2017-08-30:
why not form submissions POST and whatever's not a form submision GET?
if you are using a browser and not the captive portal using the back button becomes a pain in the ass if you navigate between POSTs

tzapu, 2017-08-30:
the change shouldn't be big either, i guess just changing the form to POST and making sure it gets the params from the POST rather than string, not sure if that part is not transparent anyway

tzapu, 2017-08-30:
like you ve actually done in your last commit :P
i'll but out, you are a machine :D

tablatronix, 2017-08-30:
I tested and changed all to post, I think it better than to have to decide which page will have a form on it etc. we can adjust as needed, if it is a problem with browser history.

tablatronix, 2017-08-30:
- [ ] test all devices with POST on all forms provide feedback on browser back and refresh capability.

tablatronix, 2017-08-30:
why not replace form buttons with actual proper links ?

tablatronix, 2017-08-30:
buttons and links styled similar
Ill probably leave this out until we actually need a redesign, I would like to add a tiny css framework to do the UI.

[image]