tzapu/WiFiManager #412: assign a custom parameter for variable

kdeklvadiya, 2017-09-01:
@tzapu 

thanks for the wonderful work done by you...... 

i have raise this issue just because i don't have enough knowledge towards programming. i am very new to programming i.e. copy-paste programmer and facing an issue with custom variable for my sketch 

Its just for measuring a +Vcc and sending an email while battery goes low. i have set battery full & battery low variable to set an alarm. somehow i managed to got those field in wifi setting page buts its not make any effect to my sketch event its not show any value.

here is my code for your review, requesting you to look into matter. i will be great full for any help to resolve my issues.

#include <FS.h>                   //this needs to be first, or it all crashes and burns...
//#define BLYNK_DEBUG           // Comment this out to disable debug and save space
#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

char blynk_token[34] = "xxxxxxxxxxxxxxxxxxxxxxxxxxxx";

bool shouldSaveConfig = false; //flag for saving data

char batt_full[5] = "2760";
char batt_low[5] = "2140";

#include <BlynkSimpleEsp8266.h>
#include <SimpleTimer.h>
ADC_MODE(ADC_VCC);
SimpleTimer timer;

unsigned int notified = 0;
void resetNotified() {
  notified = 0;
}

void battvolt() {
  float battlevel = ESP.getVcc();
  float v = 100.0 * (battlevel - batt_low[5]) / (batt_full[5] - batt_low[5]);
  Blynk.virtualWrite(V10, v);
  if (v <= 60 && notified == 0) {
    notified = 1;
    Blynk.email("xxxxxx@xxxx", "Alert", "Please recharge the battery");
    timer.setTimeout(18000L, resetNotified); // 15 min between emails
  }
}

void saveConfigCallback () {  //callback notifying us of the need to save config
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup()
{

  Serial.begin(115200);
  Serial.println();

  //SPIFFS.format();    //clean FS, for testing
  Serial.println("Mounting FS...");    //read configuration from FS json

  if (SPIFFS.begin()) {
    Serial.println("Mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("Reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("Opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          strcpy(blynk_token, json["blynk_token"]);

        } else {
          Serial.println("Failed to load json config");
        }
      }
    }
  } else {
    Serial.println("Failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 33);   // was 32 length
  WiFiManagerParameter p_batt_full("batt_full", "batt full", batt_full, 5);
  WiFiManagerParameter p_batt_low("batt_low", "batt low", batt_low, 5);

  Serial.println(blynk_token);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  wifiManager.setSaveConfigCallback(saveConfigCallback);   //set config save notify callback

  //set static ip
  // this is for connecting to Office router not GargoyleTest but it can be changed in AP mode at 192.168.4.1
  //wifiManager.setSTAStaticIPConfig(IPAddress(192,168,10,111), IPAddress(192,168,10,90), IPAddress(255,255,255,0));

  wifiManager.addParameter(&custom_blynk_token);   //add all your parameters here
  wifiManager.addParameter(&p_batt_full);
  wifiManager.addParameter(&p_batt_low);

  //wifiManager.resetSettings();  //reset settings - for testing

  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  //wifiManager.setMinimumSignalQuality();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep, in seconds
  wifiManager.setTimeout(600);   // 10 minutes to enter data and then Wemos resets to try again.

  //fetches ssid and pass and tries to connect, if it does not connect it starts an access point with the specified name
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("SmartBase", "12345678")) {
    Serial.println("Failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }
  Serial.println("Connected SmartBase :)");   //if you get here you have connected to the WiFi

  strcpy(blynk_token, custom_blynk_token.getValue());    //read updated parameters
  strcpy(batt_full, p_batt_full.getValue());
  strcpy(batt_low, p_batt_low.getValue());

  if (shouldSaveConfig) {      //save the custom parameters to FS
    Serial.println("saving config");
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    json["blynk_token"] = blynk_token;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("Failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());

  Blynk.config(blynk_token);
  Blynk.connect();

  timer.setInterval(6000L, battvolt);

}

void loop()
{
  Blynk.run(); // Initiates Blynk
  timer.run(); // Initiates SimpleTimer
}

tablatronix, 2017-09-01:
Please edit your post and use proper code
Blocks for code

kdeklvadiya, 2017-09-02:
@tablatronix 

Thanks for looking into matter....

I have try to do as you have said....I don't know weather its ok or not.

Thanks in advance.

kdeklvadiya, 2017-09-03:
@tablatronix , @tzapu 

Requesting you to look into matter whenever you get time

Many thanks in advance.

Regards
Krunal

tablatronix, 2017-09-03:
Ill try but you probably need to simplify your sketch a bit to narrow it down at first glance it looks like you are just doing it wrong, what does the example do

kdeklvadiya, 2017-09-03:
@tablatronix 
Thanks for the replay

I have tried it in many way but no change.....I have tried google but not get any matching result if you can share any example  or link then I will find someway to get the result.

Regarding code, I have copy paste from multiple example, some how its work but if you can   ##simplify it then I will grateful to you..

kdeklvadiya, 2017-09-03:
Its measure the supply voltage and send it to blynk and its send an email as an when battery voltage cross the set limit at define time interval.

tablatronix, 2017-09-03:
Yeah pull out all the blynk stuff and test

kdeklvadiya, 2017-09-03:
Actually its work fine with wifi manager I can change blynk token also but i need to change upper and lower limit of battery level through wifimanager config page.

kdeklvadiya, 2017-09-04:
@tablatronix  @tzapu 

Hi I have get the solution by trial & error. My working code is as described below. However if you can help me to make it simple and better plz share your view....I will be very grateful to you......

Thanks again for your concern and spending time for my query.....

```#include <FS.h>                   //this needs to be first, or it all crashes and burns...
//#define BLYNK_DEBUG           // Comment this out to disable debug and save space
#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

char blynk_token[34] = "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxc";

char batt_full[6] = "4000";
char batt_low[6] = "2000";

bool shouldSaveConfig = false; //flag for saving data

#include <BlynkSimpleEsp8266.h>
#include <SimpleTimer.h>
ADC_MODE(ADC_VCC);
SimpleTimer timer;

unsigned int notified = 0;
void resetNotified() {
  notified = 0;
}

void saveConfigCallback () {  //callback notifying us of the need to save config
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void battvolt(){
  float battlevel = ESP.getVcc();
  int v;
  v = 100 *(battlevel - atoi(batt_low)) / (atoi(batt_full) - atoi(batt_low));
  Blynk.virtualWrite(V10, v);
  if (v <= 60 && notified == 0) {
    notified = 1;
    Blynk.email("xxxxxxxxxx@xxxxxx", "Smart Base Alert", "Please recharge the battery");
    timer.setTimeout(900000L, resetNotified); // 15 min between emails
  }
}

void setup()
{

  Serial.begin(115200);
  Serial.println();

  //SPIFFS.format();    //clean FS, for testing
  Serial.println("Mounting FS...");    //read configuration from FS json

  if (SPIFFS.begin()) {
    Serial.println("Mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("Reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("Opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);
[... 81 lines cut ...]
    if (!configFile) {
      Serial.println("Failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());
  
  Blynk.config(blynk_token);
  Blynk.connect();

  timer.setInterval(6000L, battvolt);

}

  void loop()
{
  Blynk.run(); // Initiates Blynk
  timer.run(); // Initiates SimpleTimer  
}```

closed 2017-09-04 by kdeklvadiya