tzapu/WiFiManager #415: feature templating

tablatronix, 2017-09-08:
I would like to add basic templating , ideas on how to implement welcome.

For starters, modifying http_input string for scan rows, what is the best way ?
A callout on wifiscan, a setter, some other templating method

tablatronix, 2017-09-08:
adding wifiscan template tags
{R} RSSI
{e} encryption type

tablatronix, 2017-09-08:
Adding some styles
adding callouts,  similar to kentaylors branch

[image]

tzapu, 2017-09-09:
as a mention, there was a very nice pull request around here, using webpack to spit out all the templates needed in a header file. it was cool because it did all the minimising and uglyfying for you.

tablatronix, 2017-09-09:
I will look around, in only remember the json one but it was bloated with a personal project and jquery

tzapu, 2017-09-13:
i think this is the one
https://github.com/tzapu/WiFiManager/pull/282

thanks for all your great work

tablatronix, 2017-09-13:
Ill look it over, ignored that because I only say the translation part

jakerabid, 2018-01-08:
Would it help if I was to see if I can resolve the merge conflicts in #282 and do some testing? I suppose I could fork this and add the code I need, but that is a code deadend and I'd rather my time be useful :)

tablatronix, 2018-01-08:
This is on next milestone so not really

jakerabid, 2018-01-08:
okay, I'll fork and work against my own feature, and rework my sketch in the future once this has landed. Thanks for all your hard work @tablatronix

tablatronix, 2018-01-08:
Please post ideas also.
I thought about callout filters

jakerabid, 2018-01-08:
I'll have a think and see, but at the moment I feel like just having all the strings (along with their corresponding html) be parametrised and overridable on the server (IE on the esp8266) as in #393 would be the simplest method that would give us the smallest complied code.

I guess the other option would be to do the templating in javascript, using something like https://github.com/janl/mustache.js, or even better http://handlebarsjs.com/, then changing the content would be nice and contained.... We would be swapping processing power on the esp8266 for a larger download to the users phone but I am not sure that is really an issue.

tablatronix, 2018-01-08:
Well there is 2 things that need to happen all strings in a new include and callouts for real time logic

tablatronix, 2018-01-08:
The only problem I have with that pattern is each callback needs a a setter,  a constant, and maybe a global. Sonif we suddenly need 10 customs then its non maintainable.

tzapu, 2018-01-09:
for people that really want to customise the heck out of it, would it not be better maybe to have it somehow serve the html files from spiffs?
if it finds routes/templates on spiffs (or maybe a toggle to load from sd)load that, if not, present the embedded portal
this would allow pretty much endless customisation withour us having to worry about size of lib
it would also allow us to provide a smarter, prettier configuration portal, that is optional

also, i don't say this enough, but thank you @tablatronix , you are wonderful :D

tablatronix, 2018-01-09:
I think 
.h file const
spiffs
callouts for select pages ( save, wifiscan, etc for dynamic logic )

sorriso93, 2018-03-28:
Hi I'm looking for request for new functionality but I didn't find it. Maybe this is the appropriate discussion. Is it not possibile to foresee in addition to custom ip configuration via the webserver also the hostname?

tablatronix, 2018-03-28:
Create a new feature request issue ths

tablatronix, 2019-11-05:
I have no idea what you mean lol

dontsovcmc, 2019-11-05:
I mean to increase usability WiFiManager should remove "/wifisave" page and show connect status near button

tablatronix, 2019-11-05:
wifisave exists because we have to shut the ap down to connect, you cannot just connect to the sta in the background.