tzapu/WiFiManager #416: handleWifiSave() headless issue

suculent, 2017-09-08:
Hello, I'm receiving this weird issue on AutoConnect.

Source (line 195):
https://github.com/suculent/thinx-lib-esp8266-arduinoc/blob/9d598cc119d78f39398f15acd14c3d822c95df77/src/THiNXLib.cpp

What is my concern that the handleWifiSave() method  which:

a) seems to send WiFi save page while there is no user connected at all in the station mode...
b) does not server->close(); the connection (may not be such an issue)

Log:
```
*WM:
*WM: AutoConnect
bcn 0
del if1
mode : sta(5c:cf:7f:f0:94:9a)
mode : sta(5c:cf:7f:f0:94:9a) + softAP(5e:cf:7f:f0:94:9a)
Fatal exception 28(LoadProhibitedCause)
```

Stacktrace:
```
0x4020125a: Print::write(char const*) at ??:?
0x4020125a: Print::write(char const*) at ??:?
0x40107530: vPortFree at ??:?
0x40216f0c: wifi_set_broadcast_if at ??:?
0x40216f42: wifi_set_opmode at ??:?
0x40201344: Print::print(char const*) at ??:?
0x402033ef: ESP8266WiFiGenericClass::mode(WiFiMode) at ??:?
0x4020e343: WiFiManager::handleWifiSave() at ??:?
0x4020fba0: ESP8266HTTPUpdate::runUpdate(Stream&, unsigned int, String, int) at ??:?
0x4010020c: _umm_free at umm_malloc.c:?
0x40100688: free at ??:?
0x4024ef9f: chip_v6_unset_chanfreq at ??:?
0x4020fd15: ESP8266HTTPUpdate::handleUpdate(HTTPClient&, String const&, bool) at ??:?
0x40202f96: uart_wait_tx_empty at ??:?
0x40217b8b: wifi_station_get_connect_status at ??:?
0x402037cc: ESP8266WiFiSTAClass::status() at ??:?
0x4020483c: THiNX::connect_wifi() at ??:?
0x40204928: THiNX::connect() at ??:?
0x40204962: THiNX::connect() at ??:?
0x40214298: String::StringIfHelper() const at ??:?
0x4020175d: String::copy(char const*, unsigned int) at ??:?
0x402037cc: ESP8266WiFiSTAClass::status() at ??:?
0x4020661d: THiNX::loop() at ??:?
0x40201261: Print::write(char const*) at ??:?
0x4010020c: _umm_free at umm_malloc.c:?
0x40100688: free at ??:?
0x40214ad5: loop at ??:?
```

suculent, 2017-09-08:
I don't understand why this happens, when handleWifiSave is callback set only here:
```
server->on("/wifisave", std::bind(&WiFiManager::handleWifiSave, this));
```

tablatronix, 2017-09-08:
That is alot of stuff going on, that exception is typically an illegal memory access problem. You are gonna have to come up with a test case

suculent, 2017-09-08:
Should I use ArduinoUnit? I’m thinking about unit tests for my library as it becomes huge.

Question is, how could the handleWifiSave() happen at the first place. No captive portal client should be redirected to this without doing anything meaningful.

tablatronix, 2017-10-29:
I mean a minimal code to reproduce

closed 2018-01-26 by tablatronix