tzapu/WiFiManager #417: bssid support

tablatronix, 2017-09-08:
Allow bssid saving

- [ ] work out use interface
- [ ] thoroughly test begin with bssid saving

Instead of adding a ton of options, maybe make bssid implicit when showdupaps is true ?

tablatronix, 2017-09-08:
#221

tzapu, 2017-09-09:
test cases for reconnect with begin()
- 2 aps with same ssid/pwd, different bssid, ssid, pwd & bssid given to begin
- 2 aps with same ssid/pwd, different bssid, ssid, pwd given to begin (does it store bssid for the connected network if not given?)
- 2 aps with same ssid/pwd, different bssid, ssid, pwd & bssid given to begin, the connected bssid leaves (does it hang or reconnect to the other one even if bssid is different)

i was under the impression that bssid was not saved, but i did not check it in a looooong while

tablatronix, 2017-09-09:
I am pretty sure it works, i tested it I think, and i am in the process up getting my openwrt array up( its just 2 devices) But it will let me test ap enviroments better.

flexifi, 2019-04-23:
Does BSSID is now implemented

tablatronix, 2019-04-23:
No

flexifi, 2019-04-28:
you mean BSSID is now working sir

tablatronix, 2019-04-28:
Not in wifimanager, this issue is still open