tzapu/WiFiManager #420: Branch develop

tablatronix, 2017-09-13:
@tzapu what versioning do you think we should use for this release ?
Do we want to stay pre v1, esp8266 core is still a little unstable, 3.4.0 should be alot more so.

Let me know your thoughts, and ill update the milestones and future releases

tzapu, 2017-09-14:
Hi Shawn, it looks like you will include a massive amount of things in the next release.
I guess it merits a new major version number as well if you think that's right
Whatever you think is best i'm ok with with the mention that if there are breaking changes, it really SHOULD be a major version change

dontsovcmc, 2019-11-05:
There are huge number of features on development branch, but there is no release =(. I think, the main goal - to make main features without any customizations.

Priority:
- handlers, core optimizations
- new libraries support

Not needed for everybody:
- a lot of customizations (styles)
- ESP32 support (this is a second step)

I think the most users can't undersand what 'persist' means in you docs.., but this option has a lot of source code in the library. 

For myself: no release - no pull requests =(.

tablatronix, 2019-11-05:
The docs need to be written before I release it, I will release an alpha version to get people started and volunteering.

what do you mean no pull request?

dontsovcmc, 2019-11-05:
> what do you mean no pull request?

I mean refactoring. It's difficult, cause no releases on dev branch

tablatronix, 2019-11-05:
what refactoring ?