tzapu/WiFiManager #424: Using autoconnect to connect to WIFI, then startConfigPortal, fails to connect/save new credentials

Jsolo1, 2017-09-15:
Hi, the idea is the following, by default I'm using Autoconnect to connect to a saved wifi network, but I also want to bring up the Configuration Portal in case after pressing a button.
It connects to wifi fine, it brings up the portal fine in case I press the button, but if I select a new wifi SSID and password, it seems to ignore it, It doesn't connect to it and it doesn't save the credentials.

Any idea? Thanks

` 
setup(){
wifiManager.setConfigPortalTimeout(180);
wifiManager.autoConnect("WIFI", "password")
}
loop{
 if (portal == 1) {
    if (!wifiManager.startConfigPortal("Novaerus AQS2 WIFI", "Novaerus2017")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.reset();
      delay(5000);
    }
    else{
      Serial.println("exit with positive value");
      WiFi.mode(WIFI_STA);
    }
    portal = 0;
  }
  
}`

tablatronix, 2017-09-15:
Did you enable debug and watch serial monitor when saving?

Jsolo1, 2017-09-15:
I think I found what the problem is.
In the function :
```

int WiFiManager::connectWifi(String ssid, String pass) {
  DEBUG_WM(F("Connecting as wifi client..."));
  DEBUG_WM(WiFi.SSID());

  // check if we've got static_ip settings, if we do, use those.
  if (_sta_static_ip) {
    DEBUG_WM(F("Custom STA IP/GW/Subnet"));
    WiFi.config(_sta_static_ip, _sta_static_gw, _sta_static_sn);
    DEBUG_WM(WiFi.localIP());
  }
  //fix for auto connect racing issue
  if (WiFi.status() == WL_CONNECTED) {
    DEBUG_WM("Already connected. Bailing out.");
    return WL_CONNECTED;
  }
  //check if we have ssid and pass and force those, if not, try with last saved values
  if (ssid != "") {
	//WiFi.persistent(false);
	//WiFi.disconnect();
	//WiFi.persistent(true);
    WiFi.begin(ssid.c_str(), pass.c_str());
  } else {
    if (WiFi.SSID()) {
      DEBUG_WM("Using last saved values, should be faster");
      //trying to fix connection in progress hanging
      ETS_UART_INTR_DISABLE();
      wifi_station_disconnect();
      ETS_UART_INTR_ENABLE();

      WiFi.begin();
    } else {
      DEBUG_WM("No saved credentials");
    }
  }

  int connRes = waitForConnectResult();
  DEBUG_WM ("Connection result: ");
  DEBUG_WM ( connRes );
  //not connected, WPS enabled, no pass - first attempt
  if (_tryWPS && connRes != WL_CONNECTED && pass == "") {
    startWPS();
    //should be connected at the end of WPS
    connRes = waitForConnectResult();
  }
  return connRes;
}
```

If it's already connected, it doesn't try to connect again. Even commenting the lines:

```
 if (WiFi.status() == WL_CONNECTED) {
    DEBUG_WM("Already connected. Bailing out.");
    return WL_CONNECTED;
  }
```

Still doesn't work because WiFi.begin() won't work if there is already a connection made.
So the only solution would be to disconnect the Wifi (without deleting the credetials) and then try to connect again.

Maybe something like:

`WiFi.disconnect();`

Or

```
WiFi.persistent(false);
WiFi.disconnect();
WiFi.persistent(true);
```

Would do the trick?

Jsolo1, 2017-09-15:
Yes, it works! This is the code:

```
int WiFiManager::connectWifi(String ssid, String pass) {
  DEBUG_WM(F("Connecting as wifi client..."));
  DEBUG_WM(WiFi.SSID());

  // check if we've got static_ip settings, if we do, use those.
  if (_sta_static_ip) {
    DEBUG_WM(F("Custom STA IP/GW/Subnet"));
    WiFi.config(_sta_static_ip, _sta_static_gw, _sta_static_sn);
    DEBUG_WM(WiFi.localIP());
  }
  //fix for auto connect racing issue
  //if (WiFi.status() == WL_CONNECTED) {
  //  DEBUG_WM("Already connected. Bailing out.");
  //  return WL_CONNECTED;
  //}
  //check if we have ssid and pass and force those, if not, try with last saved values
  if (ssid != "") {
	WiFi.persistent(false);
	WiFi.disconnect();
	WiFi.persistent(true);
    WiFi.begin(ssid.c_str(), pass.c_str());
  } else {
    if (WiFi.SSID()) {
      DEBUG_WM("Using last saved values, should be faster");
      //trying to fix connection in progress hanging
      ETS_UART_INTR_DISABLE();
      wifi_station_disconnect();
      ETS_UART_INTR_ENABLE();

      WiFi.begin();
    } else {
      DEBUG_WM("No saved credentials");
    }
  }

  int connRes = waitForConnectResult();
  DEBUG_WM ("Connection result: ");
  DEBUG_WM ( connRes );
  //not connected, WPS enabled, no pass - first attempt
  if (_tryWPS && connRes != WL_CONNECTED && pass == "") {
    startWPS();
    //should be connected at the end of WPS
    connRes = waitForConnectResult();
  }
  return connRes;
}

```

The only problem are those commented lines to "fix for auto connect racing issue" Any idea what do they do?
Thanks

tablatronix, 2017-09-15:
Yeah thats a known issue, already fixed in next release.

closed 2017-09-20 by Jsolo1

tablatronix, 2017-09-20:
I think a workaround, would be to disconnect in your own code beforehand. Not sure if we can hotfix this or not