tzapu/WiFiManager #425: AP unstable, specially when there are a lot of networks

Jsolo1, 2017-09-20:
In a quite enviroment I can very easily connect to the Access point created by Autoconnect. But when there are a lot of Wifi networks around it is impossible to connect to it. Tried with multiple devices across different platforms  and none of them can manage to make it work. I found the following issue which seems very similar:

https://github.com/tzapu/WiFiManager/issues/227

Checking with a Wifi Analyzer, the wifi goes on and off:
[image]

Then I included a few lines of code before the AP is created by the WifiManager:

```
void WiFiManager::setupConfigPortal() {
  dnsServer.reset(new DNSServer());
  server.reset(new ESP8266WebServer(80));

  DEBUG_WM(F(""));
  _configPortalStart = millis();

  DEBUG_WM(F("Configuring access point... "));
  DEBUG_WM(_apName);
  if (_apPassword != NULL) {
    if (strlen(_apPassword) < 8 || strlen(_apPassword) > 63) {
      // fail passphrase to short or long!
      DEBUG_WM(F("Invalid AccessPoint password. Ignoring"));
      _apPassword = NULL;
    }
    DEBUG_WM(_apPassword);
  }

  //optional soft ip config
  if (_ap_static_ip) {
    DEBUG_WM(F("Custom AP IP/GW/Subnet"));
    WiFi.softAPConfig(_ap_static_ip, _ap_static_gw, _ap_static_sn);
  }
	WiFi.persistent(false);
    WiFi.disconnect();
	WiFi.persistent(true);
	wifi_station_set_reconnect_policy(false);
	wifi_station_set_auto_connect(false);
  if (_apPassword != NULL) {
    WiFi.softAP(_apName, _apPassword,3);//password option
  } else {
    WiFi.softAP(_apName,"",3);
  }

  delay(500); // Without delay I've seen the IP address blank
  DEBUG_WM(F("AP IP address: "));
  DEBUG_WM(WiFi.softAPIP());

  /* Setup the DNS server redirecting all the domains to the apIP */
  dnsServer->setErrorReplyCode(DNSReplyCode::NoError);
  dnsServer->start(DNS_PORT, "*", WiFi.softAPIP());

  /* Setup web pages: root, wifi config pages, SO captive portal detectors and not found. */
  server->on("/", std::bind(&WiFiManager::handleRoot, this));
  server->on("/wifi", std::bind(&WiFiManager::handleWifi, this, true));
  server->on("/0wifi", std::bind(&WiFiManager::handleWifi, this, false));
  server->on("/wifisave", std::bind(&WiFiManager::handleWifiSave, this));
  server->on("/i", std::bind(&WiFiManager::handleInfo, this));
  server->on("/r", std::bind(&WiFiManager::handleReset, this));
  //server->on("/generate_204", std::bind(&WiFiManager::handle204, this));  //Android/Chrome OS captive portal check.
  server->on("/fwlink", std::bind(&WiFiManager::handleRoot, this));  //Microsoft captive portal. Maybe not needed. Might be handled by notFound handler.
  server->onNotFound (std::bind(&WiFiManager::handleNotFound, this));
  server->begin(); // Web server start
  DEBUG_WM(F("HTTP server started"));

}

```

With no success.

Any ideas?

Thanks

tablatronix, 2017-09-20:
This is usually only an issue when you have bad sta info and the esp is trying to connect.

Have you tried erasing the config to see if it stabilizes ?

Jsolo1, 2017-09-20:
Ok, erasing the credentials makes it stable but there must be a way to stop it from trying to connect over and over, because when the ESP is moved from one network to another, where there are multiple other networks around, it would be impossible to erase/change the credentials unless connecting it to a computer.
Maybe making the function that tries to connect slower? Like every 30 seconds / 1 minute?

tablatronix, 2017-09-20:
The workaround is this

```
boolean  WiFiManager::startConfigPortal(char const *apName, char const *apPassword) {
  //setup AP
  if(!WiFi.isConnected()){
    // this fixes most ap problems, simply doing mode(WIFI_AP) does not work if sta connection is hanging
    wifioff();
    wifimode(WIFI_AP);
    DEBUG_WM("Disabling STA");
  }
  else wifimode(WIFI_AP_STA);
```

tablatronix, 2017-09-20:
```
if(!WiFi.isConnected()){
WiFi.persistent(false);
WiFi.disconnect();
WiFi.mode(WIFI_AP);
WiFi.persistent(true);
} else WiFi.mode(WIFI_AP_STA);
```

Jsolo1, 2017-09-20:
That did the trick, i've also added a small loop to try to connect 5 times before starting the portal, and a timeout in the portal with a reset, that way it's error proof!
Thanks!

closed 2017-09-20 by Jsolo1

janoist1, 2018-01-11:
I seem to be having the same issue: It is very unstable, sometimes it allows me to connect sometimes it doesn't.
It's not fully clear what was the workaround, can you explain it a bit more detailed please? Thanks.

janoist1, 2018-01-12:
Ok, I think I've got it. Putting the code above into the `configModeCallback` callback looks to have been resolved my issue. 👍

gunhansancar, 2018-03-22:
@janoist1 @tablatronix When I put that code inside `configModeCallback` it works ok but then in the Config page it doesn't show the available wifi networks. Like the `scan` function doesn't work anymore.