tzapu/WiFiManager #428: [SOLVED?]: "using" problem with RSSI and SSID

giovannicandotti, 2017-09-24:
Hi,
I faced a problem when compiling under Debian with Eclipse.
In header **ESP8266WiFi.h**

class ESP8266WiFiClass : public ESP8266WiFiGenericClass, public ESP8266WiFiSTAClass, public ESP8266WiFiScanClass, public ESP8266WiFiAPClass {
    public:
[...]

        **using ESP8266WiFiSTAClass::SSID;
        using ESP8266WiFiSTAClass::RSSI;**
        using ESP8266WiFiSTAClass::BSSID;
        using ESP8266WiFiSTAClass::BSSIDstr;

        **using ESP8266WiFiScanClass::SSID;**
        using ESP8266WiFiScanClass::encryptionType;
        **using ESP8266WiFiScanClass::RSSI;**
        using ESP8266WiFiScanClass::BSSID;
        using ESP8266WiFiScanClass::BSSIDstr;
        using ESP8266WiFiScanClass::channel;
        using ESP8266WiFiScanClass::isHidden;
[...]

causes a compiling error on **ESP8266WiFi.cpp**
/* Wifi config page handler */
void WiFiManager::handleWifi(boolean scan) {
[...]
      // old sort
      for (int i = 0; i < n; i++) {
        for (int j = i + 1; j < n; j++) {
          if (WiFi.RSSI(indices[j]) > WiFi.RSSI(indices[i])) {
            std::swap(indices[i], indices[j]);
          }
        }
      }
[...]

and later on too, involving WiFi.SSID too.
Imo, it seems due to compiler not being able to properly address to ESP8266WiFiScanClass::RSSI.

**Solution:** 
1) in **ESP8266WiFiScan.h** add something like 
        **String mySSID(uint8_t networkItem);
        int32_t myRSSI(uint8_t networkItem);**

2) in **ESP8266WiFiScan.ccp** add something like 
**int32_t ESP8266WiFiScanClass::myRSSI(uint8_t i) {	return RSSI(i);   }**
**String ESP8266WiFiScanClass::mySSID(uint8_t i) {	return SSID(i); }**

3) in **ESP8266WiFi.h** add something like 
        **using ESP8266WiFiScanClass::mySSID;
        using ESP8266WiFiScanClass::myRSSI;**

4) in **ESP8266WiFi.cpp** substitute 
**WiFi.RSSI** with **WiFi.myRSSI**
**WiFi.SSID** with **WiFi.mySSID**

Sorry for poor coding style, quick & dirty solution and maybe violation of naming convention
Just an idea to solve.

tablatronix, 2017-09-24:
do you have exact errors ?

giovannicandotti, 2017-09-24:
attached the screenshot

tablatronix, 2017-10-29:
I see no screenshot

giovannicandotti, 2017-10-30:
attached to this email

tablatronix, 2017-10-30:
You have to goto github to post images

closed 2018-01-26 by tablatronix