tzapu/WiFiManager #429: is there any way to Integrate beginSmartConfig with this library?

bkrajendra, 2017-09-25:
is there any way to Integrate beginSmartConfig with this library?

tablatronix, 2017-09-25:
Can you link to some reference for me?

bkrajendra, 2017-09-25:
Sorry for incomplete info.
Expressif has something called ESPtouch for ESP8266 and ESP32

http://espressif.com/en/products/software/esp-touch/overview

Its very easy to configure ESP8266 or ESP32 through it.

```
WiFi.mode(WIFI_AP_STA);
  delay(500);
  WiFi.beginSmartConfig();
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    Serial.println(WiFi.smartConfigDone());
  }
  Serial.println("WiFi connected");  
  Serial.println(WiFi.localIP());
```

There is simple Android Java Library provided by expressif which is very easy to use from Android App or cordova library. With this there is no need of WiFi Manager as such, But as it has its limitation we can keep both WiFi manager captive portal and SmartConfig side by side.

It should be made available on press of some button or on some event.
If it fails we can go for captive portal.

I've modified WiFimanager code for it but it does not looks like standard solution.
Its better if its implemented by someone who already known WiFiManager better.

tablatronix, 2017-09-25:
Yeah ok this is the thing that sniffs data and pairs the ap using that credential, I do not have android to test this. Does it work well, i thought it was flaky.

bkrajendra, 2017-09-27:
Works Excellent.
Its working fine for me. I tested it on ESP8266 and ESP32. Even tested it with multiple devices to send SmartConfig packets at a time.... it works.

Looks ok. People are talking about its issue with not compatible with 5GHz wifi channel.
But its good to have it as option 1 and if does not work we can go for captive portal.

tablatronix, 2017-09-27:
Ahh I see ill look at the examples

bkrajendra, 2017-09-27:
I've created this cordova plugin.
Locally i've tested it working fine. Git version not tested yet.
```
https://github.com/IOCare/cordova-plugin-smartconfig
```

tablatronix, 2017-09-27:
There seems to be a few espressif IOS apps , but they do not appear to work.

bkrajendra, 2017-09-28:
Try this. Actually its just single line but i tried to make it more useful for me.
Change include header file for ESP8266 wifi.

```
#include "WiFi.h"
unsigned long _connectTimeout         = 0;
boolean       _tryWPS                 = false;

void setup() {
  Serial.begin(115200);
  //Init WiFi as Station, start SmartConfig
  WiFi.mode(WIFI_AP_STA);
  if (connectWifi("", "") == WL_CONNECTED)   {
    Serial.println("IP Address:");
    Serial.println(WiFi.localIP());
    //connected
  }
}

void loop() {
}

int connectWifi(String ssid, String pass) {
  Serial.println("Connecting as wifi client...");

  //fix for auto connect racing issue
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("Already connected. Bailing out.");
    return WL_CONNECTED;
  }
  //check if we have ssid and pass and force those, if not, try with last saved values
  if (ssid != "") {
    WiFi.begin(ssid.c_str(), pass.c_str());
  } else {
    if (WiFi.SSID()) {
      Serial.println("Using last saved values, should be faster");
      //trying to fix connection in progress hanging
      //ETS_UART_INTR_DISABLE();
      //wifi_station_disconnect();
      //ETS_UART_INTR_ENABLE();

      WiFi.begin();
    } else {
      Serial.println("No saved credentials");
      //Added Smart config by ioCare
      WiFi.beginSmartConfig();
      //Wait for SmartConfig packet from mobile
      Serial.println("Waiting for SmartConfig.");
      while (!WiFi.smartConfigDone()) {
        delay(500);
        Serial.print(".");
      }
      //Wait for WiFi to connect to AP
      Serial.println("Waiting for WiFi");
      while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
      }

      Serial.println("WiFi Connected.");

      Serial.print("IP Address: ");
      Serial.println(WiFi.localIP());

    }
  }

  int connRes = waitForConnectResult();
  Serial.println ("Connection result: ");
  Serial.println ( connRes );
  //not connected, WPS enabled, no pass - first attempt
  if (_tryWPS && connRes != WL_CONNECTED && pass == "") {
    //startWPS();
    //should be connected at the end of WPS
    connRes = waitForConnectResult();
  }
  return connRes;
}

uint8_t waitForConnectResult() {
  if (_connectTimeout == 0) {
    return WiFi.waitForConnectResult();
  } else {
    Serial.println("Waiting for connection result with time out");
    unsigned long start = millis();
    boolean keepConnecting = true;
    uint8_t status;
    while (keepConnecting) {
      status = WiFi.status();
      if (millis() > start + _connectTimeout) {
        keepConnecting = false;
        Serial.println("Connection timed out");
      }
      if (status == WL_CONNECTED || status == WL_CONNECT_FAILED) {
        keepConnecting = false;
      }
      delay(100);
    }
    return status;
  }
}
```

tablatronix, 2017-09-28:
`if (WiFi.SSID()) {`
is wrong
`WiFi.SSID() != NULL` should work for empty config or erased config

tablatronix, 2017-09-28:
Does anyone know how airkiss works, specifically with wechat? I thought a simple ap qrcode would work , but it must require something special.

tablatronix, 2017-09-28:
Its not me, 
https://github.com/esp8266/Arduino/issues/3494

tablatronix, 2017-09-28:
Ok i tried this at home and it works, it must be only suitable for soho networks, or it just takes a really long time and the app times out or the esp crashes before it can work in environments with lots of APs, I have over 30 in my testing location,

bkrajendra, 2017-09-29:
I don't know how airkiss works. But it looks like esp has it already as there is a lib file in SDK called libairkiss.a. Or might be smartconfig is using similar library.

tablatronix, 2017-09-29:
I think it is essentially the same thing, but airkiss is built into wechat, But the docs are only in chinese, and i have no idea how you initialize the pairing ( using wechat that is )

tablatronix, 2017-09-29:
Do you have any ideas how wm would incorporate this ?
Since it only runs in STA mode, it cannot work along side WM, so not sure how this could even be beneficial unless adding the code and letting user switch it on, but then why even include it at all

dneykov, 2019-11-09:
Hello,

Any news on this? Is it possible to use smart config with WiFiManager? I would like to use it in order to configure my ESP8266 using iOS with ESP-Touch protocol.

Thanks.

tablatronix, 2019-11-09:
I can look at it again, but I don't see a reason to incorporate it, when you can just do it in code as it cannot work with wm ap mode..

bkrajendra, 2019-11-10:
Actually wm and SmartConfig are little different ideas.
While wm gives you complete config portal to configure your device through ap - just like a router, smartconfig provides different approach for device configuration - something like zeroconfig, where you dont need any config portal on device.

There is one big advantage with using wm, which makes it possible to save additional parameters by  providing a config portal.

The major challenge with wifi provisioning and device config is the tedious process which makes it difficult for end user to use the device during first time config.

What everyone wants is easy, smooth and one click process to setup wifi on device.
Using SmartConfig does address this issue by some extent, but does not work with all networks.
wm is good but makes it difficult and adds multiple steps for device configuration for end user.

We can some how combine these two advantages together by providing some settings to enable smartconfig at startup, if successful then just bypass wm config if not then enter into wm autoconfig mode. also to add some on-demand way to to start smartconfig again.

Also I'm always in favor of rest api for mobile companion app interface, I understand this is off-topic, but thats all what it comes down eventually for any commercial product.

tablatronix, 2019-11-10:
I have not found a decent smartconfig app, has the esp one been improved ?

bkrajendra, 2019-11-10:
If you work with cordova, I have this small example app.
You need to have all requirements for cordova done before following steps.
Check this out:
https://www.iocare.in/smartconfig

this will create a complete app with smartconfig functionality.

[image]

Load ESP with example smartconfig sketch.
In app scan your all wifi APs.
Select your local AP from drop down.
Enter password for it.
Click start config to configure.