tzapu/WiFiManager #431: On Demand Configuration or AutoConnect; but not both?

JSCSJSCS, 2017-09-29:
In the Documentation it says:
> On Demand Configuration Portal
> 
> If you would rather start the configuration portal on demand rather than automatically on a failed connection attempt, then this is for you.
> 
> Instead of calling autoConnect() which does all the connecting and failover configuration portal setup for you, you need to use startConfigPortal(). **Do not use BOTH**.

I have a project where the ESP8266 has to be given a distance input (how far away an object is). Because the sensor is not very accurate, it can be a trial and error to set it correctly. I was going to use this On Demand Configuration Portal to have the user press a button to enter AP configuration mode were they could re-enter the distance input (like changing MQTT server, topics, etc.), but it looks like the ESP8266 will lose it's "automatic" capabilities if I do this? Either it works automatically or you have to push a button, but not both? Is this how it works?  Maybe this is not the best approach in my use case.  Any suggestions?

tablatronix, 2017-09-29:
That means do not use both together in normal use, you can still combine them.

JSCSJSCS, 2017-09-29:
@tablatronix Thanks. Can't wait to try it.

skx, 2018-01-08:
@JSCSJSCS - Did you try?  Can this issue be closed?

closed 2018-01-26 by tablatronix