tzapu/WiFiManager #437: it does not work

shahinesi, 2017-10-11:
[image]

tablatronix, 2017-10-11:
Why not post your code or say which examples you tried

shahinesi, 2017-10-12:
@tablatronix How to do send ??

shahinesi, 2017-10-12:
[image]

shahinesi, 2017-10-12:
[image]

tablatronix, 2017-10-12:
Test using one of the simple examples, or post actual code.

shahinesi, 2017-10-13:
@tablatronix i used 
https://github.com/tzapu/WiFiManager/tree/master/examples/AutoConnect
but dose not workd!

[image]
plz help me

tablatronix, 2017-10-13:
Ststus 3 means connected, your module is connecting already.

Try erasing settings or using the ondemand example

shahinesi, 2017-10-15:
how do erasing settings??

closed 2017-10-15 by shahinesi