tzapu/WiFiManager #448: Programatically changing SSID & Password while connected?

gjt211, 2017-11-03:
I have been using this excellent library for some time now and would like to implement something that I can't find or work out a solution for.

I would like on rare occasions to change the SSID & password.

In my scenario, the ESP8266 is connected to an AP, and also to an MQTT server.
What I would like to do is send a new SSID and password, then have them stored wherever they need to go, then I can reboot the ESP8266 and it will connect to the newly stored AP.

I am not asking how to send the values, I can already do that. What I want to know is once I have the values in variables, how can I update the existing ones?

Can this be done?

tablatronix, 2017-11-03:
You can just call begin("ssid","password") using standard esp calls, and grab the ssid and password however you want, then esp.reset().

gjt211, 2017-11-03:
@tablatronix, thanks for your response.
Maybe because I just woke up or I am not overly bright, I am having trouble understanding your answer.

Nowhere in my code is the call `begin("ssid","password")` as WiFiManager handles all that for me. So I am a little confused about adding it.
When you say "grab the ssid and password" doesn't help me, I (the ESP) already knows what the ssid & password are, but I want to change them.

There is a good chance I didn't explain my problem very well. I will try again...

- The ESP is connected to (AP1) and working well
- I want to connect to a different AP (AP2)
- I know (AP1) ssid & password
- I have received (AP2) ssid & password (via MQTT)
- I now want to disconnect from AP1 and connect to AP2

As a bit of a fail-safe, my code will keep a copy of the original (AP1) ssid & password in eeprom along with a flag to say we have tried changing them. If the connect attempt to the new (AP2) fails, I will put the original (AP1) settings back and reconnect to that.

Thanks in advance.

tablatronix, 2017-11-04:
Wifimanager does not manage your sta it just calls begin using the esp stored creds

Begin saves them if persistent is true
So all you have to do it call begin with a2
And reboot

tablatronix, 2017-11-04:
Or do a wlconnected loop and fallback to old or use the multi capability built in to esp

gjt211, 2017-12-06:
Just thought I would add an update to let everyone know I have this partially working.
I am using a different project, so currently I am receiving the new SSID & PSK via the UART interface.
I have a command interpreter that receives commands as follows (in the Arduino serial terminal);
```
ssid=new_ssid_here
psk=new_password_here
```
I then execute the command
```
restartwifi
```
If this command is received, my device then runs the following code
```
WiFi.disconnect();
    yield();
    delay(100);
    WiFi.begin(configuration.my_ssid, configuration.my_psk);
    reboot = true;
    return;
```
The `reboot` flag is tested in my main loop and runs my reboot function.
The ESP8266 then reboots and attempts connection using the updated ssid/password.

_FYI: My command interpreter stores the ssid and password in EEPROM using a struct. The struct contains other stuff as well, but thats where the `configuration.my_ssid` and `configuration.my_psk` come from._

skx, 2018-01-08:
Thanks for the update @gjt211.  Since it is now working for you would you mind closing the issue?

closed 2018-12-18 by gjt211

tablatronix, 2018-12-18:
Thats a pretty good idea, I like the uart way of setting default creds, very useful for programming as you do not need to do it everytime, you could also load from spiffs if no ssid set, and use spiffs upload to upload fs automatically ( at least with platformio )