tzapu/WiFiManager #449: problems with wifi keys

alejotenis, 2017-11-03:
[ejemplo.txt](https://github.com/tzapu/WiFiManager/files/1442507/ejemplo.txt)
thank you very much for the library.

I have a problem when I access the internet through the key of my cell phone, it works perfectly, but when I log in with the key of another cell phone, this error comes out.

  ets Jan 8 2013, rst cause: 4, boot mode: (3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v4010744c
~ ld

Can anybody help me. Sorry for my English.

tablatronix, 2017-11-03:
Is that the code? Have you tried simplifying it or testing the examples?

alejotenis, 2017-12-06:
Hi. Thank you for support me.
I have some problems when I try to connecting to the wifi of cell phone, it´s just some cell phone because with my cell phone connecting perfectly but with some asus or samsung celphone it´s not working. I need connect to ubidots and thats part of programn is the problem. this is my code. The part of error are this 4 lines

` x[x1];
  ssid.toCharArray(x,x1+1);
  y[y1];
  `pass.toCharArray(y,y1+1);``

i have to do that to use ubidots.  do you know another form to connecting to ubidots with esp8266?

Thank you for your atention and sorry for my english.

[code.txt](https://github.com/tzapu/WiFiManager/files/1535203/code.txt)
[error.txt](https://github.com/tzapu/WiFiManager/files/1535204/error.txt)

closed 2018-01-26 by tablatronix