tzapu/WiFiManager #453: Encrypted key exchange

ThomasDevoogdt, 2017-11-08:
Are there plans to make the key exchange from web app to the esp8266 secure? 
I'm thinking on a Diffie-Hellman key exchange. For the moment it's extremely simple to capture the wifi password that the user has provided. 
 (https://en.wikipedia.org/wiki/Diffie%E2%80%93Hellman_key_exchange)

tablatronix, 2017-11-11:
I assume you are mostly concerned with open auth mode ?
That would require a c++ lib and a javascript library, both which are probably huge in compaison.

How about rot13 and nonces ? I mean there is no absolute secure way of client side encyption without using tls.

There has been some discussion of swapping out httpd for the async webserver, which allegedly supports tls, but this would have to be optional compile flag, since it is probably also larger.

tablatronix, 2017-11-11:
refs
https://github.com/dchest/tweetnacl-js

ThomasDevoogdt, 2017-11-12:
These are indeed the two case where I had some concerns (open auth or when using a lot of modules). You have indeed a point that these would require additional libraries which are stored on the ESP. When you are talking about rot13 or other techniques, I'm just thinking that it's smarter to not return the password in a plain URL request but as a POST payload. Because people could see it when you are configuring it. 

This weekend I adapted the library a bit so it can work with [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer). Now I'm loading the WiFiManager as an Iframe in a custom web page. And the page is hosted on the internal ESP storage. I had even success on including a video. So I think that space is in some cases no problem. Maybe It should be a possibility to store these JavaScript files there?

Here's some result.
[image]

closed 2024-04-03 by ThomasDevoogdt