tzapu/WiFiManager #455: setTimeout not working

thubaggins, 2017-11-14:
Hi,
I'm using ESP-12E with wifimanager 0.12
I try to config portal timeout to 60 seccond but it doesn't work.
`  WiFiManager wifiManager;
  wifiManager.setAPCallback(configModeCallback);
  wifiManager.setTimeout(60);
  if (!wifiManager.autoConnect("Station name", "12345678")) {
    ESP.restart();
  }`
When timeout reach wifimanager still in portal, please check this!

tablatronix, 2017-11-14:
Are any clients connecting to the AP ?

thubaggins, 2017-11-14:
No matter clients connected to ESP8266-station or not portal still doesn't restart to enter AP mode.
I just wonder why ESP-32 fork for ESP-32 work fine!

tablatronix, 2017-11-14:
esp32 branch is just a clone of master , nothing has been committed to it.

tablatronix, 2017-11-14:
I noticed sometimes if a client connects, the esp8266 does not update clientlist if they leave. I think this is fixed in the latest SDK and core. rc2, have not tested it though.

thubaggins, 2017-11-14:
No I used this fork: https://github.com/zhouhan0126/WIFIMANAGER-ESP32
Not sure what happened but maybe sdk or esp8266 hardware cause issues?

tablatronix, 2017-11-14:
I doubt it, this probably introduced a bug #308 
Your esp thinks a client is connected to the ap , and will not timeout.
Would be my guess

thubaggins, 2017-11-15:
But I have no client connected to the AP for sure!
I don't know why esp8266 just stuck in AP mode  when esp32 work fine.

tablatronix, 2017-11-15:
Yeah thats a known bug, ap persists you will have to workaround it by setting mode in your code after wifimanager is used.

thubaggins, 2017-11-15:
I’ll try it, but wonder why esp32 work when esp8266 is not!

tablatronix, 2017-11-15:
because that code was removed 
```
#if defined(ESP8266)
    if(_configPortalTimeout == 0 || wifi_softap_get_station_num() > 0){
#else
    if(_configPortalTimeout == 0){  // TODO
```

tablatronix, 2017-11-15:
If i had to guess, something is wrong with 
`wifi_softap_get_station_num()` and it is not returning 0 when no clients.

thubaggins, 2017-11-15:
Oh I see, thanks for pointing it out :)

tablatronix, 2017-11-15:
I have not seen that issue, but I did have problems with it not dropping to 0 after a client disconnected.
But you say a client is not connecting, it it possible a client is autoconnecting with saved credentials ?

thubaggins, 2017-11-15:
I’m using only one laptop connect to the AP. I’m sure my laptop doesn’t connect to the AP cause I tured off automatically connect in windows 10!
Any chance to improve  wifi_softap_get_station_num() Function?

tablatronix, 2017-11-15:
I added debugging to my dev PR, so it will output some debugging. I do not see any problem yet.

thubaggins, 2017-11-16:
Where is your PR :)

tablatronix, 2017-11-16:
Oh i thought i pasted it lol
https://github.com/tzapu/WiFiManager/pull/313

thubaggins, 2017-11-17:
Oops, your PR work like charm :)
```
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: wifi station disconnect
*WM: Connecting to saved AP
*WM: enableSTA PERSISTENT ON
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: 
*WM: WL_NO_SSID_AVAIL
*WM: wifi station disconnect
*WM: Disabling STA
*WM: Enabling AP
*WM: AccessPoint set password is VALID
*WM: 12345678
*WM: setupConfigPortal
*WM: Configuring access point with with SSID of ... 
*WM: Station-12345
12345678
*WM: SoftAP Configuration
*WM: --------------------
*WM: ssid:             Station-12345
*WM: password:         12345678
*WM: ssid_len:         13
*WM: channel:          1
*WM: authmode:         3
*WM: ssid_hidden:      0
*WM: max_connection:   4
*WM: beacon_interval:  100(ms)
*WM: --------------------
*WM: AP IP address: 
*WM: 192.168.4.1
Entered config mode
192.168.4.1
*WM: Starting Web Portal
*WM: HTTP server started
*WM: Config Portal Running, blocking, waiting for clients...
*WM: Portal Timeout In 59 seconds
*WM: Portal Timeout In 58 seconds
*WM: Portal Timeout In 57 seconds
*WM: Portal Timeout In 56 seconds
*WM: Portal Timeout In 55 seconds
*WM: Portal Timeout In 54 seconds
*WM: Portal Timeout In 53 seconds
*WM: Portal Timeout In 52 seconds
*WM: Portal Timeout In 51 seconds
*WM: Portal Timeout In 50 seconds
*WM: Portal Timeout In 49 seconds
*WM: Portal Timeout In 48 seconds
*WM: Portal Timeout In 47 seconds
*WM: Portal Timeout In 46 seconds
*WM: Portal Timeout In 45 seconds
*WM: Portal Timeout In 44 seconds
*WM: Portal Timeout In 43 seconds
*WM: Portal Timeout In 42 seconds
*WM: Portal Timeout In 41 seconds
*WM: Portal Timeout In 40 seconds
*WM: Portal Timeout In 39 seconds
*WM: Portal Timeout In 38 seconds
*WM: Portal Timeout In 37 seconds
*WM: Portal Timeout In 36 seconds
*WM: Portal Timeout In 35 seconds
*WM: Portal Timeout In 34 seconds
*WM: Portal Timeout In 33 seconds
*WM: Portal Timeout In 32 seconds
*WM: Portal Timeout In 31 seconds
*WM: Portal Timeout In 30 seconds
*WM: Portal Timeout In 29 seconds
*WM: Portal Timeout In 28 seconds
*WM: Portal Timeout In 27 seconds
*WM: Portal Timeout In 26 seconds
*WM: Portal Timeout In 25 seconds
*WM: Portal Timeout In 24 seconds
*WM: Portal Timeout In 23 seconds
*WM: Portal Timeout In 22 seconds
*WM: Portal Timeout In 21 seconds
*WM: Portal Timeout In 20 seconds
*WM: Portal Timeout In 19 seconds
*WM: Portal Timeout In 18 seconds
*WM: Portal Timeout In 17 seconds
*WM: Portal Timeout In 16 seconds
*WM: Portal Timeout In 15 seconds
*WM: Portal Timeout In 14 seconds
*WM: Portal Timeout In 13 seconds
*WM: Portal Timeout In 12 seconds
*WM: Portal Timeout In 11 seconds
*WM: Portal Timeout In 10 seconds
*WM: Portal Timeout In 9 seconds
*WM: Portal Timeout In 8 seconds
*WM: Portal Timeout In 7 seconds
*WM: Portal Timeout In 6 seconds
*WM: Portal Timeout In 5 seconds
*WM: Portal Timeout In 4 seconds
*WM: Portal Timeout In 3 seconds
*WM: Portal Timeout In 2 seconds
*WM: Portal Timeout In 1 seconds
*WM: Portal Timeout In 0 seconds
*WM: config portal has timed out
*WM: configportal abort
*WM: disconnect configportal
*WM: disconnect configportal - softAPdisconnect failed
*WM: config portal exiting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: wifi station disconnect
*WM: Connecting to saved AP
*WM: enableSTA PERSISTENT ON
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: 
*WM: WL_CONNECTED
*WM: IP Address:
*WM: 192.168.0.201
*WM: unloading

```

thubaggins, 2017-11-17:
I think I found what's happening. I've installed version 0.12 using Arduino Library Manager. In WiFiManager.cpp I found configPortalHasTimeout() was removed. After clone library from github and timeout function now working!
Diff's here: https://www.diffchecker.com/kMzI3Vet

thubaggins, 2017-11-17:
Your PR it's work!
As I said before, startConfigPortal() set WiFi mode to STA+AP cause root problem esp won't reset after timeout reach. I try to change WiFi mode in startConfigPortal() to AP only then everything working property.
```
  //WiFi.mode(WIFI_AP_STA); // Commented out this line
  WiFi.mode(WIFI_AP); // Add this line
```
Also I've add some debug message:
```
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1
*WM: SET AP
Entered config mode
192.168.4.1
*WM: 
*WM: Configuring access point... 
*WM: ESP-AP
*WM: 12345678
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Portal Timeout In 9 seconds
*WM: Portal Timeout In 8 seconds
*WM: Portal Timeout In 7 seconds
*WM: Portal Timeout In 6 seconds
*WM: Portal Timeout In 5 seconds
*WM: Portal Timeout In 4 seconds
*WM: Portal Timeout In 3 seconds
*WM: Portal Timeout In 2 seconds
*WM: Portal Timeout In 1 seconds
*WM: Portal Timeout In 0 seconds
*WM: config portal has timed out

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.0.201
```

tablatronix, 2017-11-17:
When you say it stays on , is it actually on, or is this a dup or #457 
Where AP is persistent after config, but configportal is not actually running ?

Does autoconnect ever return or exit ?

thubaggins, 2017-11-18:
Yes, I think this is a duplicate of [#457](https://github.com/tzapu/WiFiManager/issues/457)
AP won't shutdown and configportal actually not running. Change WiFi mode to AP and problem solved!
`WiFi.mode(WIFI_AP);`

tablatronix, 2017-11-18:
Thats kind of strange I wonder what the actual issue is though. Seems like it is working around one of the many bugs. The main problem i had was persistent toggling so ap mode didnt get stored in memory, softapdisconnect bugs.

tablatronix, 2017-11-18:
Also my pr only disables sta if its not connected so not sure if this is still an issue

closed 2018-01-26 by tablatronix

tablatronix, 2019-01-14:
This issue is closed and old