tzapu/WiFiManager #458: How to combine AutoConnectWithFSParametersAndCustomIP & OnDemandConfigPortal?

FarmerKJS, 2017-11-19:
Is there any way to do this? I tried setting it up like it shows in the example but I get an error.

tablatronix, 2017-11-19:
Whats the error?

tablatronix, 2017-11-19:
Whats the code

FarmerKJS, 2017-11-19:
https://pastebin.com/pUJXNTWh

It will show the portal if I connect the pin. If the pin is not connected it does this:
mounted file system
reading config file
opened config file
{"data_server":"api.thingspeak.com","update_interval":"30","api_key":"DBBRTU1YO98KYKBP","ip":"192.168.0.9","gateway":"192.168.0.1","subnet":"255.255.255.0"}
parsed json
setting custom ip from config
192.168.0.9
192.168.0.9
DBBRTU1YO98KYKBP
api.thingspeak.com
connected...yeey :)
local ip
0.0.0.0
0.0.0.0
0.0.0.0
Temperature: -196.60
connecting to api.thingspeak.com
connection failed

tablatronix, 2017-11-19:
unrelated, but this has no argument!
setMinimumSignalQuality()

tablatronix, 2017-11-19:
Works for me, whats wrong looks like its working fine.

tablatronix, 2017-11-19:
There is no autoconnect in your sketch

closed 2018-01-26 by tablatronix